omersiar/esp-rfid #452: mfrc522 and rdm6300 not reading

nightmarebd, 2021-08-15:
hi there,
im using nodemcu and rdm6300+mfrc522, wired according to esp-rfid Access Control  1.3.1 and 1.0.2. but none of my reader reads rfid cards. cant add user. can anyone help?
[image]

crs26, 2021-08-20:
double-check the connections, if you can't select a reader from the hardware settings there must be a problem with the wiring

Flo2830, 2021-08-24:
same for me, i can't add user

matjack1, 2022-03-09:
hey @nightmarebd and @Flo2830 are you using the `dev` branch? If you are still interested, please try that and use the debug mode so that on the serial port you'll see much more info and I can try helping you :) Good luck!

matjack1, 2023-03-28:
Looks like this new issue is the same: https://github.com/esprfid/esp-rfid/issues/587

I'm closing this one as I might get more feedback from the newer issue, but if someone here is still interested, please let me know.

closed 2023-03-28 by matjack1