omersiar/esp-rfid #458: wont save settings on reboot

ZeppDK, 2021-09-15:
im using a : WeMos D1 mini
i have tried flashing it with the  /bin/flash.bat, and esphome flasher, easy esp.
all tries result in the same result, if i change any settings, and click on the yellow bar in the bottom, it reboots, and resets all settings to default.

matjack1, 2022-03-09:
hey @ZeppDK not sure if you are still interested in this, but are you trying to compile the `dev` branch or the `stable`? I would recommend you trying the `dev` and possibly with the debug target as it will print more info on the serial port and I can try helping you troublshooting!

marktn, 2022-03-22:
Same issue, i have version 1.3.7, is that stable of dev? Do you have a dev download link?
This is where i download it, https://github.com/esprfid/esp-rfid/releases

marktn, 2022-03-22:
Thanks! Good news. Looks like a nice project.

marktn, 2022-03-22:
> @marktn try getting the binaries from here: https://github.com/esprfid/esp-rfid/suites/5709560102/artifacts/188342329
> 
> We'll soon release a new version with all the fixes! I hope in a month or so :)
@matjack1 It works!

matjack1, 2022-03-22:
Yes! Thank you for the feedback!!

closed 2022-03-22 by matjack1