omersiar/esp-rfid #459: MFRC522 doesn't recognize any RFID Tags.

zLeqitEclipse, 2021-10-06:
Hi, i have a Problem which i cant solve by myself.
I already tried anything, from chaning the antenna gain to using other RFID Chips that are confirmed to be working, but nothing helps.
No matter what i do, it just dosn't recoginze any type of RFID Chip.
I plugged every pin in like in the README.md.
Unfortunately, the log doesn't show any Communication Error or similar.

Hardware Info:

- Wemos D1 Mini (ESP8266)
- RFID MFRC-522

[image]

Any help would be appreciated.

matjack1, 2022-03-09:
hey @zLeqitEclipse not sure if you still need help, but have you tried using the `dev` branch of this project and compiling with the debug mode? You get a lot more info on the serial port and I can try helping you troubleshoot.

zLeqitEclipse, 2022-03-09:
> hey @zLeqitEclipse not sure if you still need help, but have you tried using the `dev` branch of this project and compiling with the debug mode? You get a lot more info on the serial port and I can try helping you troubleshoot.

Well no not rellay as is already have returned them long ago. But I still appreciate the help!

closed 2022-03-09 by matjack1