omersiar/esp-rfid #462: Compatiblility with ESP32 and the PN7150 NFC tag reader + "FPC BM-Lite" fingerprint reader used in "Chhavi" (crowdfunding campaign product on on Crowd Supply)

Hedda, 2021-10-19:
Do you think that the NXP PN7150 module based NFC tag reader and  "FPC BM-Lite" fingerprint sensor used in this ESP32-based "Chhavi" product can be supported by the ESP RFID project?

https://www.crowdsupply.com/vicharak/chhavi

@akshar001 from Vicharak has launched "Chhavi" ESP32 based fingerprint and NFC reader product as a crowdfunding campaign on Crowd Supply. 

"Chhavi is a compact, wireless, touch-capacitive fingerprint sensor based on ESP32 WiFi and Bluetooth SoC that is offered with optional NFC connectivity and a battery." 

It says that it can also optionally be powered by a battery.

Not cheap but it have to say that its very compact size makes it looks very nice. BTW, I first read about it on CNX-Software blog:

https://www.cnx-software.com/2021/10/08/chhavi-esp32-fingerprint-sensor-nfc-battery/

"FPC BM-Lite" fingerprint sensor module by Fingerprints 

https://www.fingerprints.com/technology/hardware/modules/fpc-bm-lite/

https://www.fingerprints.com/solutions/access/bm-lite-development-kit/

NXP "PN7150 NFC controller"

https://www.nxp.com/products/rfid-nfc/nfc-hf/nfc-readers/high-performance-nfc-controller-with-integrated-firmware-for-smart-devices:PN7150

https://www.nxp.com/docs/en/data-sheet/PN7150.pdf

#### NFC Specifications & RF Protocols
 - PN7150 NFC controller
 - NFCIP-1 ¹ & NFCIP-2 ²
 - ISO/IEC 14443A, ISO/IEC 14443B PICC, and NFC Forum T4T modes via host interface ³
 - NFC Forum T3T via host interface
 - ISO/IEC 14443A, ISO/IEC 14443B PCD designed according to NFC Forum digital protocol T4T platform and ISO-DEP ⁴
 - FeliCa PCD mode
 - MIFARE Classic PCD encryption mechanism (MIFARE Classic 1K/4K)
 - NFC Forum tag 1 to 5 (MIFARE Ultralight, Jewel, Open FeliCa tag, MIFARE DESFire) ¹
 - ISO/IEC 15693/ICODE VCD mode ⁵
 - Supports card emulation as well as reader mode

¹ ISO/IEC 18092 (NFCIP-1) edition, 2013-03-15. This is similar to Ecma 340
² ISO/IEC 15693 part 2: 2nd edition (2006-12-15), part 3: 1st edition (2001-04-01)
³ ISO/IEC 14443 parts 2: 2001 COR 1 2007 (2007-11-01), part 3: 2001 COR 1 2006 (2006-09-01) and part 4: 2nd edition 2008 (2008-07-15)
⁴ NFC Forum Device Requirements V1.3
⁵ ISO/IEC 15693 part 2: 2nd edition (2006-12-15), part 3: 1st edition (2001-04-01)

For those that want to know, the PN 7150 offloads NFC library code onto the controller so it does not have to run on the ESP. Also, it has about 150th the power usage of the PN 532, so if you can run it on a properly designed ESP32 board with a slow polling interval then should be able to operate on battery alone, though as a battery-operated wireless IoT sensor there would be some latency to get on Wi-Fi unless your WiFi has been designed to handle such devices.

PS: I'm just a home automation enthusiast and have no affiliations or connections with Chhavi and Crowd Supply or any others.

Hedda, 2021-10-19:
FYI, @akshar001 posted to ESPHome discussion that they also have plans to make a less expensive NFC-only model as well:

https://github.com/esphome/esphome/discussions/2464

matjack1, 2022-03-09:
@Hedda I'm closing this as we don't have the hardware, but we are very open to a PR whenever someone wants to do it. Thank you for the suggestion though!

closed 2022-03-09 by matjack1

Hedda, 2022-03-10:
> @Hedda I'm closing this as we don't have the hardware, but we are very open to a PR whenever someone wants to do it. Thank you for the suggestion though!

@akshar001 (Akshar Vastarpara) from Vicharak would probably send you a prototype or sample if you ask him directly, see:

https://www.crowdsupply.com/vicharak/chhavi/ask-question

Hedda, 2022-03-10:
By the way, also check out his esphome custom component repo for chhavi:

https://github.com/akshar001/chhavi_esphome

I believe it makes use of the NFC_LIB library/driver for pn7150 from his chhavi repo:

https://github.com/akshar001/chhavi

https://github.com/akshar001/chhavi/tree/main/Firmware/Arduino/NFC_LIB/