omersiar/esp-rfid #463: How hard would be the ESP32 porting?

sinclairfr, 2021-11-02:
Hi guys, 

I just broke my foot and I have a lot of time on my hand.
I am a bit of a jack of all trades, I can code and have some electronics basics.

My guess is, since ESP8266 and ESP32 share a lot of features, the porting should not be so had, should it? 

Of course, there will be pin remapping to do but apart from that, I have to confess nothing comes to mind.

Can any of you guys give me advice and/or clues in order to port this great project to ESP32?

Thanks!

AGordiGuerrero, 2021-12-24:
I would also love to do it. But I do not know how to begin. Apart from "Start new project..." :-)

We can try it together if you want.

matjack1, 2022-02-23:
Hello @sinclairfr, I hope you are well by now :)

I think the porting to ESP32 is a long standing issue, which I've briefly looked into. I've recently upgraded ArduinoJson to the v6, which I think works better with ESP32, and I'm working to better handle WiFi connections, which I think is the main issue now to support ESP32.

So I think it could come out relatively easily once the WiFi changes are done. Meanwhile, something that would be useful is understanding what is not working on the ESP32 so that we can understand if we need to maintain both platforms, or if we can switch to libraries that can manage both platform for us. Maybe we find out that there are more areas to upgrade and the porting will be more difficult, but I don't know yet :)

I'm available in Gitter: https://gitter.im/esp-rfid/Lobby if you want to have a chat.

pvtex, 2024-06-14:
I have made a port to ESP32 but only for Wiegand readers.
it is running an the web UI is working.
WIll connect a reader today and test it.

you can find it here:
https://github.com/pvtex/esp32-rfid

pvtex, 2024-06-15:
jut uploaded the first release for the esp32-rfid version 1.1.0
this is only for wiegand

https://github.com/pvtex/esp32-rfid/releases/tag/1.1.0

matjack1, 2024-06-24:
Thank you @pvtex for letting us know about that! I'm surely going to have a look and see if we can plan an ESP32 porting for the next version. Maybe this summer I will find some time to close V2 for real and then I can consider having a look at that, thank you!

tctlrd, 2025-02-04:
We are also working with the esp32 and ethernet w/PoE based on @pvtex work.
A big thank you to @matjack1 all the devs including @pvtex for this project!

We are attempting to implement MIFARE DESFire NFC support.

For now we are using the Olimex ESP32-POE2 board we may eventually move on to the RP2040/RP2350 platform since we do not use WiFi. https://wiznet.io/products/evaluation-boards

https://github.com/tctlrd/esp32-actl/tree/dev

matjack1, 2025-02-04:
hey @tctlrd have you forked again from @pvtex ? Do you think we can try joining forces to have a codebase that works both for ESP8266 and ESP32 as well?

I don't have much time, but I can review pull requests for sure and I can test with the hardware that I have :)

tctlrd, 2025-02-05:
@matjack1

1. My single repo [esp32-actl](/tctlrd/esp32-actl) is a fork from @pvtex's repo [esp32-rfid](/pvtex/esp32-rfid) which is a fork from [esp-rfid](/esprfid/esp-rfid).

2. Yes I am happy to discuss joining forces for a common codebase.

Having a codebase for multiple SoCs is a good idea. It may work best to have a separate repository for each SoC (and likely for certain board/hardware combos). These can pull the shared code from a common codebase repo. The main reason for deviding things up, is that there are different library and logic requirements depending on SoC and other hardware arrangements (i.e. ethernet, IO/relay I2C expansion boards).

3. I appreciate your enthusiasm to bring others into this.

matjack1, 2025-02-23:
I was thinking to separate as much as possible with #IFDEFs and a little bit of abstraction where necessary. I know that the libraries are different, but maybe not too much to make it impossible with my simplistic approach?

I don't have time to start, but if you want to start somewhere we could also think about merging progressively the pieces, we don't need to have everything working for ESP32 from day zero.

If anyone wants to help with ESP32 porting, what I would do is:
1. setup a different build with libraries for ESP32
2. add IFDEFs to compile only for ESP8266
3. progressively add code for ESP32 so that we can add features step by step and test along the way

If anyone wants to help let me know, I'm also open to do some work together over a call, but I don't have time to do everything myself :)

tctlrd, 2025-03-10:
@matjack1 
Thank you for sharing in detail what you are thinking!
I will discuss this with my team and we may be able to get on a call.
Our fork for esp32 is functional on Olimex ESP32-POE2: https://github.com/tctlrd/esp32-actl

matjack1, 2025-03-10:
If you want to get on a call, let me know, I'm surely interested to learn more about what you are doing, thanks!

tctlrd, 2025-03-18:
@matjack1 we will talk about it on Thursday and get back to you. Thanks for keeping on this!