omersiar/esp-rfid #464: Compiling error.

CentauriDK, 2021-11-07:
Hi I have some trouble comping this project.
I have just tried reinstalling Visual Code and Platform IO all with the newest version.
But I get 37 errors when trying to compile this project.
Is there any special version of the different libraries that have to be specified manually or something since it now working right out of the box.

I get the following errors.
And hir as image. 
[image]

`[{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonObject.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'ArduinoJson::JsonVariant::as<long long int>() const'",
	"startLineNumber": 283,
	"startColumn": 46,
	"endLineNumber": 283,
	"endColumn": 46
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 150,
	"startColumn": 3,
	"endLineNumber": 150,
	"endColumn": 3
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 156,
	"startColumn": 3,
	"endLineNumber": 156,
	"endColumn": 3
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 165,
	"startColumn": 3,
	"endLineNumber": 165,
	"endColumn": 3
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, const char*>'",
	"startLineNumber": 175,
	"startColumn": 3,
	"endLineNumber": 175,
	"endColumn": 3
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 183,
	"startColumn": 3,
	"endLineNumber": 183,
	"endColumn": 3
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, ArduinoJson::JsonArray&>'",
	"startLineNumber": 198,
	"startColumn": 3,
[... 203 lines cut ...]
	"severity": 4,
	"message": "'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]",
	"startLineNumber": 7,
	"startColumn": 12,
	"endLineNumber": 7,
	"endColumn": 12
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/src/wsResponses.esp",
	"owner": "cpp",
	"severity": 4,
	"message": "'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]",
	"startLineNumber": 17,
	"startColumn": 13,
	"endLineNumber": 17,
	"endColumn": 13
},{
	"resource": "/c:/Users/kim/Desktop/ESP_RFID_Demo/esp-rfid/src/wsResponses.esp",
	"owner": "cpp",
	"severity": 4,
	"message": "'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]",
	"startLineNumber": 52,
	"startColumn": 7,
	"endLineNumber": 52,
	"endColumn": 7
}]`

joelgamal, 2021-11-21:
try `ArduinoJson` version 5.13.4 and arduino core 2.x.x not 3.x.x

sinclairfr, 2021-12-13:
I'm having the same issues, so many errors, how did they manage to do a release for version 1.3.x with all those errors?

omersiar, 2021-12-13:
These errors weren't exist three years ago.  Luckily platformio can be configured to download specific version of all dependencies. 😊

zgibsonTFCC, 2021-12-14:
I am having the same issue. It looks like platformIO is set to use the ArduinoJson library version 5.13.4 but I am not too sure what else needs to be changed here. Any other feedback or advice would be greatly appreciated.

sinclairfr, 2021-12-14:
what helped me is running 
`platformio run -t clean` and then `platformio run`
and also make sure that `ArduinoJson@5.13.4` is written `ArduinoJson @ 5.13.4`
For some weird reason that did the trick on my mac running VSCode

sinclairfr, 2021-12-14:
also` #define ARDUINOJSON_USE_LONG_LONG 1` in main.cpp file

AGordiGuerrero, 2021-12-24:
I do not manage to compile. After including the ArduinoJson@5.13.4 and including the #define, I still have a lot of errors related with it. For example:

_`lib/ArduinoJson/src/ArduinoJson/JsonObject.hpp:283:46: error: no matching function for call to 'ArduinoJson::JsonVariant::as<long long int>() const'`_

_lib/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:150:3: error: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>_

I am running it on Ubuntu 20.04.2 with PlatformIO Core, version 5.2.3. Python 3.8.10, although I have also installed Python 2.7.18. Now is using Python3 (as said by python --version). 

¿Can anyone help? How can I use arduino core 2.x.x instead of 3.x.x?

AGordiGuerrero, 2021-12-24:
Wow! I compile it using the old Espressif core. Changing the `platformio.ini` file putting:

`platform = `espressif8266@2.6.3`

matjack1, 2022-02-23:
Were you trying to build the `stable` branch or the `dev` branch? Unfortunately I think that `dev` is much better off now, can you please confirm?

@omersiar what would you think would be needed to release a new version of esp-rfid? I think that `dev` is in much better place than `stable` now?

marktn, 2022-03-22:
HI,

I also having troubles, and see some errors.

also tried to change the platform = espressif8266@2.6.3

Errors:
Cannot find "c:\*****\ESP Rifd\esp-rfid-dev\esp-rfid-dev\include".

'MFRC522::MFRC522(byte)' is deprecated (declared at .pio\libdeps\generic\MFRC522\src\MFRC522.h:337): use MFRC522(byte chipSelectPin, byte resetPowerDownPin) [-Wdeprecated-declarations]

'deprecated' attribute directive ignored [-Wattributes]

Is it possible to get some help? The project looks very good. 
The flash.bat tool does not work at all.

matjack1, 2022-03-30:
I'm going to close this for now as I think there's no more interest, but please reopen if you need any help

closed 2022-03-30 by matjack1

fdiazf19, 2022-05-12:
Hello, is there any way to compile the  `stable` branch? I've been trying the `dev` one but there is many bugs related to Access Log and adding new users compared to `stable` branch compiled files.
This happens when i use a card in the reader, instead of showing the typical add user tab:
[image]

Access Logs shows saved cards (with Access type set as "Always") as access disabled and just for holding it a moment close to the reader produce many log access. This is an example of using (just for a second) one of the cards assigned to a user:
[image]

As you can see this user had previously been registered:
[image]

i'm using NodeMcu v3, MCRC-522 reader and VSCode. I have not modified the code yet.
Thanks in advance.

matjack1, 2022-06-29:
hey @fdiazf19 have you tried to install the new branch on an esp with data created with the stable branch? Maybe that's why is not working properly with existing data?

The stable doesn't build anymore because the platfom and the libraries didn't have specific versions specified and so now it's fetching the latest ones and doesn't build.

Unfortunately I don't know which version was working the last time and soon we'll release a new stable version, so maybe you can try getting 1.x releases from here: https://github.com/esprfid/esp-rfid/releases as binaries directly?