omersiar/esp-rfid #466: Card UID ED0BA996 read as EDBA996

windy54, 2021-11-30:
We are using version 1.3 off of the dev branch, sorry I can't be more exact, in our Hackspace.
Apart from the occasional issue it works.
I have got to be honest and say I am looking at alternatives that store the valid UID in our Hackspace management system.

All of the valid ID's captured by the ESP system have been transferred into the management system.

I am using an MFRC522 reader, an old Raspberry PI model B and python to read a card and validate it .
Some cards failed to validate and it appearr to be because the ESP system reads the UID as EDBA996, whereas the raspberry PI system reads it as ED0BA996.
An extra zero has been inserted.
To determine whether it is an issue with the PI software, I installed a sketch on an Arduino UNO to dump the card information to the serial monitor. This dumps the UID for the same card as ED0BA996 , so I do not see an issue with the PI software.

I am going to add some debug statements into our code to see if I can determine what is happening.
It is almost as if it a little/big endian problem, the actual UID is B ED 96 A9, the PI and Arduino systems add the leading 0 and the ESP does not.

Has anyone else seen this issue?

Steve

windy54, 2021-11-30:
I have looked at the code and in rfid.esp line 25
there is this loop
`for (int i = 0; i < mfrc522.uid.size; ++i)
		{
			uid += String(mfrc522.uid.uidByte[i], HEX);
		}`
I am guessing that a byte 0B gets converted to B, because leading zeroes are ignored, not an endian problem.
Can anyone please point me at the documentation for the String function, I keep searching but cannot find it.

This may be an "issue" that you do not want to fix, because all users would have to re-validate their RFID cards, possibly something to be aware of if RFID cards are being read from different systems

hinkma, 2022-01-13:
I can working this way.

uid += String(mfrc522.uid.uidByte[mfrc522.uid.size - (i + 1)] < 0x10 ? "0" : "");
uid += String(mfrc522.uid.uidByte[mfrc522.uid.size - (i + 1)], HEX);

windy54, 2023-01-19:
So i had not seen hinkma's fix and did this

`byte nextByte = mfrc522.uid.uidByte[i];
		if ( nextByte < 16){
			uid+="0"; // assuming that a byte 0A gets converted to A, i.e. 0 drops
		}
		uid += String(nextByte, HEX);`

inside the for loop, but Hinkma's solution which amounts to the same is more elegant!

matjack1, 2023-03-28:
Hey @hinkma @windy54 I've added your fix here: https://github.com/esprfid/esp-rfid/pull/586 if you want to test it :)

Thanks!

windy54, 2023-03-28:
Will do, might be a few daysSteveSent from my iPhone

matjack1, 2023-08-23:
Merged in dev branch, I'm closing this, thank you again!

PR #586 (2023-08-23): Fix mfrc522 read
Fix #466

Thank you @hinkma and @windy54!
closed 2023-08-23 by matjack1