omersiar/esp-rfid #467: PN532 - DO NOT use GPIO15 on Wemos D1 Mini

hellebauer, 2021-12-02:
The recommended connection for PN532 suggests to use GPIO15 (D8) for SS.

The problem is that GPIO15 is also used for Reset on Wemos D1 Mini boards.  As a result Power-on Reset and Reset-button no longer worked  on my Wemos, while the PN532 SS-signal was hooked up to GPIO15.

I rewired the SS to GPIO16 (D0) on my Wemos and all was well afterwards. 
_(Don't forget to set SS=GPIO16 in the esprfid user interface)._

[image]

matjack1, 2022-03-09:
@hellebauer should we write this somewhere in the documentation, do you think?

BaconWithThat, 2023-01-11:
Just found this and fixed my Wemos D1 mini that wouldn't boot when connected to the PN532, thank you.

BarrMo, 2023-01-26:
Similar with my ESP8266 dev board. Set and connected to GPIO15 just seemed to jam it up. Wifi wasn't available either. I had to connect SS on the PN532 to GPIO5 (D1) Because GPIO16 (D0) is connected to the onboard LED.

matjack1, 2023-03-28:
I hope that the new doc here: https://github.com/esprfid/esp-rfid/pull/589 sort everything out in the documentation side

PR #589 (2023-03-28): Fix doc for PN532
Fix #467
closed 2023-03-28 by matjack1