omersiar/esp-rfid #468: RC522, Wemos D1 Mini, NodeMCU V2 (ESP12E)

makertirol, 2021-12-24:
Ich versuche die ganze Zeit den RC522 mit einenWemos D1 Mini bzw. NodeMCU V2 zu verbinden aber er liest mir keinen RFID Chip aus.
Muss ich dafür einen Mqtt Broker laufen haben oder geht das auch ohne Mqtt?
Hab in den Issues etwas gelesen von  GPIO Pin 16 statt 15, aber es nützt alles nichts.
Hab auch versuchsweise einen PN532 probiert, immer das selbe.
Welche bin Datei muss ich beim NodeMCU verwenden, denn wenn ich die boardV2.bin drauf flashe dann lässt sich nichts einstellen (wir ein kleines Stopzeichen angezeigt). Hab auch schon andere Version probiert, vergebens.
Brauch ich dazu wirklich einen MqttBroker?
Wer kann mir bitte helfen.
Vielen Dank im voraus!

makertirol, 2021-12-25:
Jetzt hab ich auch noch einen Mqtt Broker eingerichtet, funktioniert aber trotzdem nicht. Bin am verzweifeln. :-(

AchimL, 2021-12-25:
Hallo makertirol,
MQTT wird erste einmal nicht benötigt.
Bei mir gab es auch einige (selbstgemachte) Probleme.
- Kabel vertauscht
- Erreichbarkeit des D1 über Wlan usw.

Als Software habe ich diese Version aufgespielt
**https://github.com/esprfid/esp-rfid/releases/tag/1.0.2**
Alle Anderen funktionierten bei mir nicht so richtig.

**Vorher habe ich den Speicher mit blank4mb.bin gelöscht, kurz vom Strom getrennt und dann o. Version aufgespielt.**

Dann kannst Du die Oberfläche über die IP 192.168.4.1 erreichen.
Kennwort: "admin".
Hier folgendes einstellen:
- Reader Type :   MFRC522
- SPI SS:             GPIO-15
- Antenna Gain: Max (48dB)

Hier bei Bedarf auch ewas einstellen (das sind meine Einstellungen für ein Torschloss):
- WiFi LED:         None
- Relay Type:      Active High
- Relay Pin:        GPIO-4
- Lock Type:       Momentary
- Activate Time:  400
- Button Pin:      None

Wenn alles richtig eingestellt ist, müsste beim "Kartenvorhalten" vor den Leser unter der Rubrik "User" ein Eintrag erzeugt werden, der nur noch zugeordnet werden muss.
**Also wie eingangs beschrieben, erst einmal die eigenen Hardwareverbindungen prüfen**
Erst wenn die Kartenerkennung funktioniert, den Wemos in das eigene Wlan-Netz integrieren und MQTT einschalten.

Viel Erfolg,
Achim

makertirol, 2021-12-26:
Hallo Achim,
Danke für deine Hilfe. Habe den D1 komplett gelöscht mit der Arduino Software,
danach die Ver.1.0.2 neu aufgespielt, alles eingestellt, den MFRC522 neu verkabelt.
Leider wieder keine Funktion. Hab nun einen anderen MFRC522 angeschlossen
und siehe da, es funktioniert. Hätte mir nie gedacht das dieser defekt sei, war neu.
Naja kann auch bei neuem Gerät passieren, kostet ja nichts ☺️
Vielen Dank nochmal für deine Hilfe.

AchimL, 2021-12-26:
Prima!
Manchmal steckt der Wurm drin!

Achim

matjack1, 2022-02-23:
I cannot read German, but this looks closed to me :)