omersiar/esp-rfid #469: 37 bit wiegand support

scienkoptic, 2021-12-28:
I have a ESP_RFID Blue unit that I was planning to use with HID Proxpoint 6005 reader. The ESP-RFID will detect handle a 26 bit card, but I am trying to specifically read a 37 bit card. The reader reads the 37 bit card, as the red LED changes to green and the reader emits a beep. I do not have any additional hardware to test the reader with, but I have three of them and they all behave the same way. 

Does the ESP-RFID support 37 bit format cards? 
_

matjack1, 2022-01-17:
hey @scienkoptic you can try changing this line here: https://github.com/esprfid/esp-rfid/blob/dev/src/rfid.esp#L7 to read the 37bit cards. I don't have such a reader to test, so I'm not 100% sure it will work.

Moreover, try different GPIO ports as depending on the device some of them might not work, but I haven't understood why. So just try different combinations before giving up!

scienkoptic, 2022-01-19:
Thank you so much for your help. I'm not very proficient with github. The copy I downloaded as a zip file contains a different version of the rfid.esp. Is there a method for downloading the version you reference as a whole?

scienkoptic, 2022-01-19:
Hi, I made this change, but also needed to update magicnumbers.h and wiegand.h library for it work.
the changes I made were to search for and replace all instances of 34 in the three files with 37. perhaps I got lucky. 

So thank you for giving me the impetus to try.

closed 2022-01-19 by scienkoptic