omersiar/esp-rfid #470: Wiegand reader leds and buzzer control.

hinkma, 2022-01-13:
Would it be possible for the Wiegand reader to get outputs for LEDs, green and red, and a buzzer? 
From an accepted reading, the green LED would light up for the set time and the buzzer would give a uniform beep. If there was no right of way then the red LED would light up for the set time and the buzzer would give an intermittent beep.

matjack1, 2022-01-17:
hey @hinkma can you share some documentation of your specific wiegand device? I've tried understanding on mine if it's possible to manage lights and buzzer but I couldn't find any docs. It would be interesting!

hinkma, 2022-01-17:
I use HID multiCLASS SE reader. It is Red and Green light and buzzer own input. When input connect GND (down) it activates the action. This normal almost every Wiegand in the commercial reader. https://www.hidglobal.com/doclib/files/resource_files/plt-01579-a.4_iclass_se_multiclass_se_installaton_guide.pdf

CountParadox, 2022-02-24:
Happy to donate an hid iclass se reader to the Dev if cover post

matjack1, 2022-04-06:
hey @hinkma and @CountParadox we have implemented recently support for beeper and LED for access denied or for pincode waiting:

[image]

you can wire them to the wiegand led/beeper, right?

hinkma, 2022-04-08:
Hi. I can test weekend. Thanks.

hinkma, 2022-04-08:
Can some one build ESP12E program?
I can try build program and give next fault:

[{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonObject.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'ArduinoJson::JsonVariant::as<long long int>() const'",
	"startLineNumber": 283,
	"startColumn": 46,
	"endLineNumber": 283,
	"endColumn": 46
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 150,
	"startColumn": 3,
	"endLineNumber": 150,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 156,
	"startColumn": 3,
	"endLineNumber": 156,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 165,
	"startColumn": 3,
	"endLineNumber": 165,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, const char*>'",
	"startLineNumber": 175,
	"startColumn": 3,
	"endLineNumber": 175,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 183,
	"startColumn": 3,
	"endLineNumber": 183,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, ArduinoJson::JsonArray&>'",
	"startLineNumber": 198,
	"startColumn": 3,
	"endLineNumber": 198,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, const ArduinoJson::JsonArray&>'",
	"startLineNumber": 208,
	"startColumn": 3,
	"endLineNumber": 208,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, ArduinoJson::JsonObject&>'",
	"startLineNumber": 219,
	"startColumn": 3,
	"endLineNumber": 219,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, const ArduinoJson::JsonObject&>'",
	"startLineNumber": 230,
	"startColumn": 3,
	"endLineNumber": 230,
	"endColumn": 3
},{
	"resource": "/c:/Users/Marko/Desktop/RFID/esp-rfid-2.x/esp-rfid-stable/esp-rfid-stable/.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/JsonVariant.hpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'",
	"startLineNumber": 238,
	"startColumn": 3,
	"endLineNumber": 238,
	"endColumn": 3
},{

hinkma, 2022-04-08:
I can build, my mistake. Problem in Platformio

hinkma, 2022-04-10:
First thing, I use dev  branch, I can't build and upload new program to ESP. ESP stuck boot loop at the start. I build new program and transfer that by webui update when older version is installed , then new program start.
Second things, logs not working locally, maybe MQTT, I am not tested jet.
Reader LEDs and buzzer work properly. 
Two cosmetic things, if 'Pin code requested' is false, can green LED illuminate same time than buzzer give access ok signal. If 'Pin code requested' is true and system waiting PIN code, can green LED blinking?