omersiar/esp-rfid #479: change html page

omaxali, 2022-02-08:
Hello every one.
can any one describe step-by-step how we could change sth in the first page in html code?using gulp.
for example changing the name of menus. or adding some descriptions.
Please###Please###Please###
thanks in advance.

matjack1, 2022-02-08:
@omaxali have a look at the documentation here: https://github.com/esprfid/esp-rfid/blob/dev/README.md#frontend

I'm trying to restore the executables so that you don't need to install NodeJS etc, but right now you need to follow the steps there.

omaxali, 2022-02-09:
> 

thank you for your rapid reply.
I installed the last version of noedjs and
I followed the steps, i get an error while executing npm start
[image]
is there anything wrong with the nodejs or gulp version?

omaxali, 2022-02-16:
> > 
> 
> thank you for your rapid reply. I installed the last version of noedjs and I followed the steps, i get an error while executing npm start [image] is there anything wrong with the nodejs or gulp version?

any help?
@matjack1

matjack1, 2022-02-22:
yes @omaxali sorry I've been a bit busy :)

Please have a try with the binaries that you can find here: https://github.com/esprfid/esp-rfid/actions/runs/1883814716 and let me know if they work. You should copy the binary in the folder where you are trying to build the web interface, so in `tools/webfilesbuilder` and run it from there, with a double click too, it should do the job :)

omaxali, 2022-02-25:
@matjack1
Thank you SOooooooooo much.
Magical :) 💯 :)

closed 2022-02-25 by omaxali

rodrigodc299, 2022-10-22:
Hy, same error.
I installed the last version of noedjs and
I followed the steps, i get an error while executing npm start
https://user-images.githubusercontent.com/50978042/153131805-59989fe4-8ca4-4a9d-bb69-6730cbbc2a21.png
 Can you help me Pleaseeee?

matjack1, 2022-11-21:
hey @rodrigodc299 sorry for the delay. I don't think you need to install node if you are using the executables.

Either you use the raw executables or you use nodejs. If you use nodejs directly you should go to the `tools/webfilesbulider` folder in the terminal, install the dependencies with `npm install` and run `npm start` to bundle everything.

Let me know if that works.

With the executable instead, remember to copy it in the `tools/webfilesbuilder` folder.