omersiar/esp-rfid #486: Rc522 8266 open web operation random crash restart

vvvyky, 2022-03-04:
Rc522 8266 open the web operation, randomly crash and restart, and SPI is connected to miso-12 mosi-13 sck-14 sspin-15. I've tried many 12F modules. How can I solve them?

[image]

Exception (3):
epc1=0x40101638 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4005a839 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffecb0 end: 3fffffb0 offset: 01a0
3fffee50:  ffffffff 00000000 3ffeb7a1 00000008  
3fffee60:  40269026 3ffeede8 3fff337c 00000001  
3fffee70:  00000094 00000000 00000020 4010182a  
3fffee80:  000000c0 4025751e 00000008 3fff5f8a  
3fffee90:  40268090 3ffeede8 00000000 40101110  
3fffeea0:  00000000 40257687 3fff3624 40272660  
3fffeeb0:  00000000 00000002 00000001 00000008  
3fffeec0:  3fff337c 3ffef018 3ffef018 402548a8  
3fffeed0:  40254e4f 3ffef018 3ffef018 40254dfa  
3fffeee0:  00000018 00000001 00000001 3ffede8a  
3fffeef0:  3fff5f6c 000000ba 3ffede9a 3fff3744  
3fffef00:  3ffede74 00000000 00000021 40268596  
3fffef10:  3ffef018 3ffede74 3fffdcc0 3ffeb6b8  
3fffef20:  3ffede84 3fff507c 00000000 3ffeb6b8  
3fffef30:  00000000 3ffef018 00000000 3ffe85f8  
3fffef40:  40267e6f 3fffdab0 00000000 4021e76f  
3fffef50:  3ffeb6b8 40000f49 3fffdab0 40000f49  
3fffef60:  40000e19 40001878 00000002 00000000  
3fffef70:  3fffff10 aa55aa55 000000ed 40105741  
3fffef80:  40105747 00000002 00000000 534a0054  
3fffef90:  4010000d 00470400 0054004d 00480000  
3fffefa0:  00000000 3fffef3c 00000000 3ffffc48  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  feefeffe feefeffe feefeffe feefeffe  
3ffff150:  feefeffe feefeffe feefeffe feefeffe  
3ffff160:  feefeffe feefeffe feefeffe feefeffe  
3ffff170:  feefeffe feefeffe feefeffe feefeffe  
3ffff180:  feefeffe feefeffe feefeffe feefeffe  
3ffff190:  feefeffe feefeffe feefeffe feefeffe  
3ffff1a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff1f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff200:  feefeffe feefeffe feefeffe feefeffe  
3ffff210:  feefeffe feefeffe feefeffe feefeffe  
3ffff220:  feefeffe feefeffe feefeffe feefeffe  
3ffff230:  feefeffe feefeffe feefeffe feefeffe  
3ffff240:  feefeffe feefeffe feefeffe feefeffe  
3ffff250:  feefeffe feefeffe feefeffe feefeffe  
3ffff260:  feefeffe feefeffe feefeffe feefeffe  
3ffff270:  feefeffe feefeffe feefeffe feefeffe  
3ffff280:  feefeffe feefeffe feefeffe feefeffe  
3ffff290:  feefeffe feefeffe feefeffe feefeffe  
3ffff2a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff2f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff300:  feefeffe feefeffe feefeffe feefeffe  
3ffff310:  feefeffe feefeffe feefeffe feefeffe  
3ffff320:  feefeffe feefeffe feefeffe feefeffe  
3ffff330:  feefeffe feefeffe feefeffe feefeffe  
3ffff340:  feefeffe feefeffe feefeffe feefeffe  
3ffff350:  feefeffe feefeffe feefeffe feefeffe  
3ffff360:  feefeffe feefeffe feefeffe feefeffe  
3ffff370:  feefeffe feefeffe feefeffe feefeffe  
3ffff380:  feefeffe feefeffe feefeffe feefeffe  
3ffff390:  feefeffe feefeffe feefeffe feefeffe  
3ffff3a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff3b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff3c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff3d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff3e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff3f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff400:  feefeffe feefeffe feefeffe feefeffe  
3ffff410:  feefeffe feefeffe feefeffe feefeffe  
3ffff420:  feefeffe feefeffe feefeffe feefeffe  
3ffff430:  feefeffe feefeffe feefeffe feefeffe  
3ffff440:  feefeffe feefeffe feefeffe feefeffe  
3ffff450:  feefeffe feefeffe feefeffe feefeffe  
3ffff460:  feefeffe feefeffe feefeffe feefeffe  
3ffff470:  feefeffe feefeffe feefeffe feefeffe  
3ffff480:  feefeffe feefeffe feefeffe feefeffe  
3ffff490:  feefeffe feefeffe feefeffe feefeffe  
3ffff4a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff4b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff4c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff4d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff4e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff4f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff500:  feefeffe feefeffe feefeffe feefeffe  
3ffff510:  feefeffe feefeffe feefeffe feefeffe  
3ffff520:  feefeffe feefeffe feefeffe feefeffe  
3ffff530:  feefeffe feefeffe feefeffe feefeffe  
3ffff540:  feefeffe feefeffe feefeffe feefeffe  
3ffff550:  feefeffe feefeffe feefeffe feefeffe  
3ffff560:  feefeffe feefeffe feefeffe feefeffe  
3ffff570:  feefeffe feefeffe feefeffe feefeffe  
3ffff580:  feefeffe feefeffe feefeffe feefeffe  
3ffff590:  feefeffe feefeffe feefeffe feefeffe  
3ffff5a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff5b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff5c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff5d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff5e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff5f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff600:  feefeffe feefeffe feefeffe feefeffe  
3ffff610:  feefeffe feefeffe feefeffe feefeffe  
3ffff620:  feefeffe feefeffe feefeffe feefeffe  
3ffff630:  feefeffe feefeffe feefeffe feefeffe  
3ffff640:  feefeffe feefeffe feefeffe feefeffe  
3ffff650:  feefeffe feefeffe feefeffe feefeffe  
3ffff660:  feefeffe feefeffe feefeffe feefeffe  
3ffff670:  feefeffe feefeffe feefeffe feefeffe  
3ffff680:  feefeffe feefeffe feefeffe feefeffe  
3ffff690:  feefeffe feefeffe feefeffe feefeffe  
3ffff6a0:  feefeffe feefeffe feef�?)�������
[ INFO ] NFC_WIFI_CAR_KEY v1.3.5

matjack1, 2022-03-06:
hey @vvvyky sorry I cannot undertand how you get the restart, is by browsing the web interface? If so, can you please explain in details the steps you are taking?

Are you using the `dev` branch of the project? `stable` is now obsolete :(

I'm working on something to fix the restarts on the web UI, but it's going to happen on the `dev` branch.

vvvyky, 2022-03-07:
> hey @vvvyky sorry I cannot undertand how you get the restart, is by browsing the web interface? If so, can you please explain in details the steps you are taking?
>
> Are you using the `dev` branch of [...]

If it is development, clicking the page, button, refresh and other operations may crash randomly and restart the module

matjack1, 2022-03-07:
Can you try to send a video recording showing the steps you take to reproduce the error? I cannot understand what's going on, sorry :(

vvvyky, 2022-03-11:
> 您能否尝试发送一段视频记录，显示您为重现错误所采取的步骤？我无法理解发生了什么，对不起:(

https://user-images.githubusercontent.com/38814385/157799649-c0e76f93-574a-459a-98f1-6189718e9674.mp4

Tried a variety of modules will be like this

vetruk, 2022-03-11:
Try using d16  as  "SS" pin not D15 since is used in module boot mode. In my practice i never use it as described in documentations. Or put some 10k resistor to ground on pin"D15". Make sure you have sufficient power supply 1A+. I thin there is open issue specialy for not using D15 pin as "SS"

vvvyky, 2022-03-11:
> Try using d16 as "SS" pin not D15 since is used in module boot mode. In my practice i never use it as described in documentations. Or put some 10k resistor to ground on pin"D15". Make sure you have sufficient power supply 1A+. I thin there is open issue specialy for not using D15 pin as "SS"

D16 has been tried as both SS and D15 pull-down, and the same problem occurs randomly

Exception (3):
epc1=0x4010146d epc2=0x00000000 epc3=0x00000000 excvaddr=0x4002c0a0 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffea80 end: 3fffffb0 offset: 01a0
3fffec20:  00002570 000004ae 000004ae 40101570  
3fffec30:  0000000f 00000000 3fff80d4 3fffeca0  
3fffec40:  00000012 3fff552c 00000020 401017fb  
........................................................

vvvyky, 2022-03-11:
> Can you try to send a video recording showing the steps you take to reproduce the error? I cannot understand what's going on, sorry :(

Clicking the switch page, especially loading the user list and log list, is more prone to problems

vvvyky, 2022-03-11:
https://user-images.githubusercontent.com/38814385/157817239-2754fe4d-79b1-4c46-a3f2-4a452e5f962c.mp4

vvvyky, 2022-03-11:
https://user-images.githubusercontent.com/38814385/157818388-a37032df-548e-4cbf-99cd-ebdb98d9362a.mp4

matjack1, 2022-03-22:
OK, I see @vvvyky can you please try again with the binaries that you can find here: https://github.com/esprfid/esp-rfid/suites/5709560102/artifacts/188342329

I'm aware of these problems and hopefully soon we'll release a stable version with all the fixes. The linked version above should be already stabler than before though.

Let me know!

vvvyky, 2022-03-23:
> OK, I see @vvvyky can you please try again with the binaries that you can find here: https://github.com/esprfid/esp-rfid/suites/5709560102/artifacts/188342329
>
> I'm aware of these problems and [...]

The connection has been tested, but there is still a problem. Different ESP modules have the same crash and restart problem

stupsi099, 2022-03-25:
I use the file above forV2Board.bin with original Board and did full erase and flash with rs232. I got web interface and can conect to Wlan - but only with DHCP. I change to fix IP did reboot and DHCP again (at webinterface it was static IP). And if i try to read a badge it open the interface but there is no UID !!!!!  Back to latest version and all is working again. Testet on 2 from 3 original Boards with same result.

matjack1, 2022-03-27:
@stupsi099 what is the latest version that you are mentioning that works correctly?

stupsi099, 2022-03-27:
I use the official 1.3.7 version- this works. The problem is only sometime webrefresh wont work and after loose wifi it wont reconnect. And the one from the link with only dhcp and no uid

matjack1, 2022-03-28:
ok, thank you for the feedback @stupsi099

matjack1, 2023-01-22:
hey @vvvyky can you please check if this PR: https://github.com/esprfid/esp-rfid/pull/577 fixes your problem here?

@stupsi099 if you want me to have a look at your problem, can you please share you settings by exporting the settings.json file from the web interface?

For now I'm closing this issue, but please reopen if necessary. Thank you

closed 2023-01-22 by matjack1