omersiar/esp-rfid #493: Compiling error

paoloros97, 2022-03-29:
Before starting to work on a custom version, I tried to compile the code as it is.
after hitting the `platformio run` command, at the end it cames out the following error:

`.pio/libdeps/generic/ArduinoJson/src/ArduinoJson/Deserialization/../JsonVariant.hpp:238:3: error: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, long long int>'
Compiling .pio/build/generic/lib5aa/AsyncMqttClient/AsyncMqttClient.cpp.o
Compiling .pio/build/generic/lib5aa/AsyncMqttClient/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
*** [.pio/build/generic/src/main.cpp.o] Error 1
========================================================= [FAILED] Took 7.12 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
generic        FAILED    00:00:07.124
=================================================== 1 failed, 0 succeeded in 00:00:07.124 ===================================================
`
How can I fix it?

grahamsingle, 2022-03-30:
Same problem - JasonVarian.hpp:
201:  //
202:  // const JsonArray& as<const JsonArray&> const;
203:  template <typename T>
204:  typename Internals::EnableIf<
205:      Internals::IsSame<typename Internals::RemoveReference<T>::type,
206:                        const JsonArray>::value,
207:      const JsonArray &>::type                                                <=========== PROBLEM HERE
208:  as() const {                                                                
209:    return variantAsArray();
210:  }

Compiler reports: .pio\libdeps\generic\ArduinoJson\src/ArduinoJson/Deserialization/../JsonVariant.hpp:208:3: error: no type named 'type' in 'struct ArduinoJson::Internals::EnableIf<false, const ArduinoJson::JsonArray&>'

matjack1, 2022-03-30:
hey @paoloros97 and @grahamsingle it looks like you have the wrong version of ArduinoJson installed. Can you please try cleaning the target that you are compiling and trying again?

This might happen if you were on `stable`, got the old version of ArduinoJson, then moved to `dev`. Platformio doesn't get automatically the new version and happens what you are experiencing.

Let me know if how it goes!

matjack1, 2022-03-31:
@paoloros97 once you've downloaded the project, you should switch to the `dev` branche.

So, clone the project, then:

`git checkout dev`

And then try again the compilation and let me know!

paoloros97, 2022-04-01:
Thank you @matjack1 it nicely worked!
[image]
:)

matjack1, 2022-04-01:
YES! Thank you for the feedback ;)

closed 2022-04-01 by matjack1