omersiar/esp-rfid #505: Latest dev release not responding to MQTT messages.

windy54, 2022-05-04:
I have built the latest dev release and have no errors.

My MQTT broker is installed on a Raspberry PI.
I can see messages the RFID system is transmitting in the broker window and also on the serial interface from the esp-rfid.

The esp-rfid does not respond to any received messages.
Following screenshot shows the settings,

my broker is at 192.168.4.45 and the esp-rfid at 192.168.4.49

The following command 

mosquitto_pub -h 192.168.4.45 -t maindoor -m '{"doorip":"192.168.4.49:,"cmd":"opendoor"}'

is captured in a window monitoring MQTT messages but nothing is output from the esp-rfid either to the broker or on the serial link.
Broker output:
{"type":"heartbeat","time":1651689771,"uptime":8149,"ip":"192.168.4.49","hostname":"esp-rfid"}
Client (null) sending PINGREQ
Client (null) received PINGRESP
Client (null) sending PINGREQ
Client (null) received PINGRESP
Client (null) received PUBLISH (d0, q0, r0, m0, 'maindoor', ... (42 bytes))
{"doorip":"192.168.4.49:,"cmd":"opendoor"}
Client (null) sending PINGREQ

Because I have built a debug release I would expect to see some output due to the following

`void onMqttMessage(char *topic, char *payload, AsyncMqttClientMessageProperties properties, size_t len, size_t index, size_t total)
{
	size_t n = 0;
	size_t i = index;
#ifdef DEBUG
	Serial.print("[ INFO ] onMqttMessage: ");
	
#endif
	while(n < len) {
		mqttBuffer[i] = payload[n];
		n++;
		i++;
	}
	if(index + len == total) {
		mqttBuffer[i] = '\0';
	} else {
		return;
	}
#ifdef DEBUG
	Serial.print("[ INFO ] JSON msg: ");
	Serial.println(mqttBuffer);
#endif
`
I added the “OnMqttMessage” debug line.

I thnik it was a clean build, i.e. no previous RFID builds in my environment.

It must be something I am doing, but I cannot see what!
Any ideas as to what I can try ?

Steve Gale

giutrec, 2022-06-15:
you have to send the commands on the cmd queue and he writes on the send queue

giutrec, 2022-06-15:
the dev I've been using for months and it works perfectly, it doesn't respond to you because you write the command on the wrong queue, try this:

mosquitto_pub -h 192.168.4.45 -t maindoor/cmd -m '{"doorip":"192.168.4.49:,"cmd":"opendoor"}'

and the same goes for the messages he writes go to the send sub-queue

example: mosquitto_aub -h 192.168.4.45 -t maindoor/send

windy54, 2022-06-22:
Adding /cmd as suggested above solves this issue so it can be closed.
Now go to look at getuserlist, it seems to stop after 11 users but does not crash

windy54, 2022-10-11:
I have finally managed to test it and you are correct, thanks for response

Steve