omersiar/esp-rfid #507: Hardware purchase

gorstj, 2022-05-21:
Hi,

Does anyone know if there are any alternative hardware suppliers?

The Tindie page linked (the original or the 'blue' board) don't seem to ship. I have purchased but then I am refunded as it hasn't shipped after a few weeks. The seller isn't answering emails.

Any idea of suitable alternatives
(don't think I could make my own with ESp8266 board as there are two many voltage changes 3.3<-->12v for my poor skills to cope with!)

Thanks for any advice.

henfri, 2022-06-19:
Marelab.org uses a fork of esprfis and sells boards.

matjack1, 2022-10-22:
@nardev are you shipping new boards?

nardev, 2022-10-22:
Yes

matjack1, 2022-10-22:
Excellent!

So whoever is interested can buy from @nardev a custom board here: https://www.tindie.com/stores/nardev/ !

closed 2022-10-22 by matjack1

nardev, 2022-10-22:
You can ask for customization