omersiar/esp-rfid #518: Code mixes tabs and spaces

mtfurlan, 2022-06-21:
I noticed that there is a mix of tabs and spaces.
I'm happy to make a PR to clean that up, but which way should it go?
I would personally prefer spaces, but I'll make it consistent either way.

mtfurlan, 2022-06-23:
What about
 ```
 if (stuff)
 {
     ...
 }
 ```
 vs
 ```
 if (stuff) {
     ...
 }
 ```
 ?
 I thought there was only the former but I just found some of the latter.
 
Would you be interested in having an [uncrustify](http://uncrustify.sourceforge.net/) config?

I dunno what the best way to keep the codebase in line with a config would be though.
At $work I don't allow a PR to merge if it doesn't pass an automated run of the linter, but that seems like it might be overly complicated here.

matjack1, 2022-10-22:
As I mentioned here: https://github.com/esprfid/esp-rfid/pull/524#issuecomment-1179301580 I don't mind which way you prefer, so long as it's automatically set up by VS Code and ideally in the CI, so that we cannot break it again :) Thanks!