omersiar/esp-rfid #530: How to reboot ESP-RFID via MQTT in Home Assistant?

DivanX10, 2022-09-15:
I went through the topics and did not find an answer to my question, how to reboot ESP-RFID via MQTT in Home Assistant? I tried it and it didn't work out

* '{ "send":"restart"}'
* '{ "send":"restart", "doorip":"192.168.1.48"}'
* '{ "cmd":"restart"}'
* '{ "cmd":"restart", "doorip":"192.168.1.48"}'

```
service: mqtt.publish
data:
  topic: esprfid
  payload_template: '{ "cmd":"restart", "doorip":"192.168.1.48"}"
```

IvanIzquierdoJimenez, 2022-09-16:
Hi, in the mqtt.esp no include restart how cmd command, yo need add this command on .esp file

DivanX10, 2022-09-16:
> Hi, in the mqtt.esp no include restart how cmd command, yo need add this command on .esp file

Thanks, but I can't do it. If only the author can add to the firmware. I do not know how to add these commands to mqtt.esp and how to upload it to nodemcu later.

IvanIzquierdoJimenez, 2022-09-16:
You need VScode and platformio to compile and modify the code, if you need something help, tell me

DivanX10, 2022-09-16:
Yes, it would be great if you could show me what and how. For example, what code to add to the mnt.esp file and how to assemble it later, or I will be grateful for the finished firmware with the built-in command to reboot ESP-RFID

IvanIzquierdoJimenez, 2022-09-16:
Yes, I will see how to make this, and tell you my progress.

evazzoler, 2022-09-27:
IMHO it would be better if you put a home assistant relay in front of esp-rfid. That way you'll be sure to reboot esp-rfid even when/if it is totally frozen.

DivanX10, 2022-11-05:
I express my gratitude to Aleks130699 for the working esp-rfid firmware. A similar topic was raised here [FIx Home Assistant integration ](https://github.com/esprfid/esp-rfid/pull/504 ). Now you can reboot the ESP board, there are sensors displaying the label. You can download the firmware from this [link](https://drive.google.com/drive/folders/15WgpG_04JhWDdBxhBjB9pIWSCzyGV4ll ). In the settings, you need to enable MQTT for Home Assistant, after which esp-rfid will appear in the Home Assistant.

[image]
[image]
[image]

Thanks to this, I now have an excellent reading from the label. More details about my project can be found [here](https://youtu.be/71JAbj5qM5I ). Where I have embedded tags in air freshener cylinders in a smart air freshener and when changing the cylinder, the Home Assistant understands which cylinder is installed and takes into account the resource of the cylinder consumption

DivanX10, 2022-11-05:
> IMHO it would be better if you put a home assistant relay in front of esp-rfid. That way you'll be sure to reboot esp-rfid even when/if it is totally frozen.

Now this is not required, you can just download the firmware from this [link](https://drive.google.com/drive/folders/15WgpG_04JhWDdBxhBjB9pIWSCzyGV4ll) and esp-rfid will appear in the Home Assistant

closed 2022-11-05 by DivanX10

DivanX10, 2022-11-07:
A working reboot command via the MQTT service for this [firmware](https://drive.google.com/drive/folders/15WgpG_04JhWDdBxhBjB9pIWSCzyGV4ll)

```
service: mqtt.publish
data:
  topic: esprfid/cmd
  payload: "{cmd:'reboot',doorip:'192.168.1.48'}"
```

evazzoler, 2022-11-07:
I still don't understand the usefulness of restarting a working and responding device.
Please, erudite me!

matjack1, 2022-11-07:
@DivanX10 let us know what's your use case for this as I agree with @evazzoler, I cannot understand the purpose. Seems like we are trying to bypass fixing an error that you are experiencing? Would be nice to solve the root cause of the issue, rather than rebooting it to hide the problem.

DivanX10, 2022-11-07:
> I still don't understand the usefulness of restarting a working and responding device. Please, erudite me!

I have a smart air freshener that works according to the Zigbee protocol and on batteries, and I am also in the process of creating a base for an air freshener, which will have a built-in MC RC522, wemos mini reader module and power is supplied to charge batteries in it, similar to a radiotelephone that stands at a telephone exchange. This air freshener refreshes the air in the toilet through both Voice Assistant and Home Assistant automation. For a variety of flavors, I often change the cylinders, I can change them two or three times a week, or I can change them once a week, but the flavors change often. In order to know the consumption resource of the balloon flavors and when the balloon resource is coming to an end, the Home Assistant will inform me in advance that it's time to buy a new balloon. You can know which is the most frequent fragrance used. Further, the selected fragrance is automatically set in the Home Assistant according to the read label and continues accounting for this balloon further. When you change the balloon, the MC RC522 tag reader does not respond to the tag, either falls asleep, or freezes and only a reboot helps. After restarting, the balloon is read and the voice assistant informs that the balloon is installed and informs which flavor is selected. I added a reboot to the automation when I take the balloon out of the air freshener and after that the reader reads the label 100% and there is no such situation that after installing the balloon in the air freshener, the automation does not work out to the end, because the reader does not respond to the label and I had to reboot the board by de-energizing

<details>
  <summary>Photos of the Zigbee smart air freshener and a 3D model of the future base</summary>
  
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
  
</details>

<details>
  <summary>Example of automation Toilet: Air freshener. Counter. Cylinder resource</summary>
  
```
alias: "Toilet: Air freshener. Counter. Cylinder resource"
description: >-
  With each spray, a counter will be triggered and, depending on the selected
  cylinder, the calculation will be carried out for the selected cylinder. If the cylinder is
  removed, the general counter is reset, and the cylinder counters are reset
  strictly manually. If you select the desired cylinder from the drop-down list, the
  number of sprays will be restored, provided that the counter of the selected
  cylinder has not been reset manually
trigger:
  - platform: state
    entity_id:
      - sensor.tualet_osvezhitel_vozdukha_action
    attribute: state_l1
    id: Pressing the button
    alias: Pressing the button
  - platform: state
    entity_id:
      - input_select.air_freshener_replaceable_balloons
    id: The balloon is selected
    alias: The balloon is selected
  - platform: state
    entity_id:
      - input_select.air_freshener_replaceable_balloons
      - sensor.toilet_air_freshener_status
    to: The balloon is removed
    id: The balloon is removed
    alias: The balloon is removed
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id: Pressing the button
            alias: Pressing the button
        sequence:
          - choose:
              - conditions:
                  - condition: template
                    value_template: >-
                      {{ not is state('input_select.air_freshener_replaceable_balloons', 'balloon is removed') }}
                  - condition: template
                    value_template: >-
                      {{ not is_state('sensor.toilet_air_freshener_status','balloon is removed') }}
                sequence:
                  - service: counter.increment
                    data: {}
                    target:
                      entity_id: counter.toilet_air_freshener_spray_counter
[... 65 lines cut ...]
                sequence:
                  - service: counter.configure
                    data:
                      value: >
                        {% set ballon_installed = states('input_select.air_freshener_replaceable_balloons') %}
			{% if ballon_installed == 'The balloon is removed' %} 0
			{% elif ballon_installed == 'Ocean Oasis' %}
			{{ states('counter.balloon_aroma_ocean_oasis') }}
			{% elif ballon_installed == 'Peony and juicy berries' %}
			{{ states('counter.balloon_aroma_peony_and_juicy_berries') }}
			{% elif ballon_installed == 'Freshness of linen' %}
			{{ states('counter.balloon_aroma_freshness_of_linen') }}
			{% elif ballon_installed == 'Freshness of the morning' %}
			{{ states('counter.balloon_aroma_freshness_of_the_morning') }}
			{% elif ballon_installed == 'Japanese Garden' %}
			{{ states('counter.balloon_aroma_japanese_garden') }}
			{% endif %}
                    target:
                      entity_id: counter.toilet_air_freshener_spray_counter
            default: []
    default: []
mode: restart

```
</details>

evazzoler, 2022-11-07:
Ok, your project is really amazing, really original and useful.
But there are some "but", IMHO:

- your are not using esp-rfid for the purpose for which it was intended

- esp-rfid is not intended to be powered by a battery 

- you are rebooting the esp8266 for solving a problem that may exist in the reader and not in the board

- you are not fixing a issue but get around it, risking to create another issue

- you are wasting the small ESP memory with code implementing a not needed reboot command instead of fixing existing code

I believe that this way of acting does not respect the best practices of software philosophy and software engineering.

My advice is to find out why the player is misbehaving, fix the problem, remove the reboot function to make room for more useful functions.  If the internal watchdog is not sufficient, the keep-alive via MQTT must be used by an external controller to perform a power cycle of the whole device through a relay. 
For battery powered board you can use a relay with radio remote control. The only reason to reboot should be because esp-rfid no longer responds, at which point the reboot can only be hardware.

DivanX10, 2022-11-07:
> Ok, your project is really amazing, really original and useful. But there are some "but", IMHO:
>
> * your are not using esp-rfid for the purpose for which it was intended
> * esp-rfid is not intended to [...]

I don't use esp-rfid to power the battery. Everyone has their own food. The air freshener is powered by disposable batteries, which last for 3 months.
[image]

With the new firmware, I haven't rebooted esp nodemcu v3 for two days and it's too early to talk about it, but it seems to be working fine. I checked other tags and no matter how the freezes were noticed, you need to wait at least a week to understand whether it will hang or not. I did the reboot because there is no trust in the stable operation of the MC-RC522 module and there is no understanding who is the culprit, the firmware or the MC-RC522 module. It may be that the MC-RC522 module itself falls asleep, since there is no frequent access to it and when you bring the label after 2-4 days or 2 weeks, then through the esp rfid web interface I do not observe that the reader reads the tags.

matjack1, 2022-11-08:
Hey @DivanX10 but then, why don't you use the existing feature?

[image]

Wouldn't that solve your problem?

DivanX10, 2022-11-08:
> 

This restarts every 24 hours, and the hang occurs before 24 hours have passed and this option does not save the situation

matjack1, 2022-11-08:
But didn't you say before that 2 days have passed and everything works fine?

How often would you like to reboot?

DivanX10, 2022-11-08:
> But didn't you say before that 2 days have passed and everything works fine?
> 
> How often would you like to reboot?

it's only been two days with the new firmware and so far everything is fine, today I updated my Home Assistant and the last state of the sensors went wrong. While I'm conducting surveillance. I didn't attach any importance to this before, until I noticed that the reader stops working. Now I will pay attention to this, but so far too little time has passed to claim that the firmware is working correctly.

Aleks130699, 2022-11-09:
@matjack1 prior to this, @DivanX10 used the very first version 1.0.2, perhaps the hanging error was in it, as well as my assumption that this may happen due to the fact that the label is constantly near the reader, and this may cause problems

evazzoler, 2022-11-09:
> @matjack1 prior to this, @DivanX10 used the very first version 1.0.2, perhaps the hanging error was in it, as well as my assumption that this may happen due to the fact that the label is constantly near the reader, and this may cause problems

And that can fry the label or at least shorten its life...

matjack1, 2022-11-18:
@DivanX10 how is it going? Instead of the MQTT reboot, you can have a look at this: https://github.com/esprfid/esp-rfid/pull/552 :)

DivanX10, 2022-11-19:
Hello. When I checked 4 days later, it fell off, I won't say exactly on what day. Sensors were unavailable in the Home Assistant. I rebooted the board via the web console and the reader started working again.

DivanX10, 2022-11-19:
I see that you have fixed something there. The firmware file itself from where can I download it?

matjack1, 2022-11-20:
@DivanX10 you can try the latest dev build from here: https://github.com/esprfid/esp-rfid/suites/9369310704/artifacts/441298681

Try the auto-reboot functionality, it should work now :)

DivanX10, 2022-11-21:
Thanks. Good. I will try and I need a few days to check to see how the reader works with esp rfid

DivanX10, 2022-11-22:
> @DivanX10 you can try the latest dev build from here: https://github.com/esprfid/esp-rfid/suites/9369310704/artifacts/441298681
> 
> Try the auto-reboot functionality, it should work now :)

I filled in the firmware. The device did not appear in Home Assistant, although the esprfidhomeassistant topic appeared in mqtt explorer, you probably forgot to add Discovery.  I rolled back to my firmware and the device appeared in the Home Assistant

[image]

Also, if I lean the labels, then I don't see the names of the labels in the topics
[image]

Aleks130699, 2022-11-22:
This error has been corrected in my PR #549, which has not yet been accepted for work.

Aleks130699, 2022-11-22:
@matjack1 Regarding my PR, I really don’t know why support for several relays was once introduced and how it can be applied in practice, personally, in principle, I no longer have free ports that could somehow be used, I finalized their support  just because it was made

matjack1, 2022-11-22:
The fixed autorestart is now merged on the `dev` branch :)

evazzoler, 2022-11-23:
> @matjack1 Regarding my PR, I really don’t know why support for several relays was once introduced and how it can be applied in practice, personally, in principle, I no longer have free ports that could somehow be used, I finalized their support just because it was made

Multiple relays can be used to open different doors with different badges and a single reader (offices, corridors, etc.), or to enable some badges to deactivate an alarm system and others only to open the door, etc.
It is a solution that makes esp-rfid more versatile and adaptable to the needs of users who would otherwise be forced to choose commercial products.

matjack1, 2022-11-23:
After our chat I'm convinced it's useful, it just need to be checked around as sometimes the assumption is that multiple relays are driving the same "door".

DivanX10, 2023-01-03:
In general, things are like this. The RFID operation is still falling off and I abandoned this option in favor of the solution of determining the balloon on the esp. To determine the balloon, I use this Resistance Sensor component. It works stably and reliably

To understand how it works, you can watch the [video](https://youtu.be/WkUgiq72C0c) and the [documentation](https://community.home-assistant.io/t/air-refreshener-esphome-wi-fi-controlled/251205/12?u=divanx10) itself

evazzoler, 2023-01-03:
> In general, things are like this. The RFID operation is still falling off and I abandoned this option in favor of the solution of determining the balloon on the esp. To determine the balloon, I use [...]

Ok but, please, keep considering that this project is oriented on access control and access control does not require the token to be left in front of the reader. Abnormal behavior in this usage is not inherent to the project.

nikola-75, 2023-02-13:
> Выражаю благодарность Aleks130699 за рабочую прошивку esp-rfid. Похожая тема поднималась здесь [Интеграция FIx Home Assistant](https://github.com/esprfid/esp-rfid/pull/504) . Теперь можно перезагрузить плату ESP, есть датчики отображающие метку. Скачать прошивку можно по этой [ссылке](https://drive.google.com/drive/folders/15WgpG_04JhWDdBxhBjB9pIWSCzyGV4ll) . В настройках нужно включить MQTT для Home Assistant, после чего в Home Assistant появится esp-rfid.
>
Good afternoon. I tried to upgrade from firmware 1.3.3 rdm 6300 stopped working, was connected to rx (GPIO 3). Also, the keys (rc522) that were in memory stopped working, they are read once, but the lock is not opened.

matjack1, 2023-03-11:
hey @nikola-75 sorry, but this issue is for something else. If you have troubles upgrading, can you please share your settings by exporting the file from the web UI? So I can try reproducing your setup and make the upgrade path possible

nikola-75, 2023-03-12:
> привет@nikola-75Извините, но этот вопрос о другом. Если у вас возникли проблемы с обновлением, не могли бы вы поделиться своими настройками, экспортировав файл из веб-интерфейса? Поэтому я могу воспроизвести ваше население и сделать возможный путь обновления.

Good afternoon. This configuration works on the version "esp-rfid-1.3.7-dev-patchAlex130699", when updating to the version "esp-rfid-2.0.0-dev-patchAlex130699-v.1" rdm6300 falls off, rs 522 responds, but does not read memory keys. Below in the archive is the configuration, the password for the web interface is kolr220908.

[Downloads.zip](https://github.com/esprfid/esp-rfid/files/10952113/Downloads.zip)