omersiar/esp-rfid #532: Pushing a whole user list or retrieving via URL

danielschonfeld, 2022-10-08:
In some installations, it might be less brittle to not sync one user at a time via MQTT, but rather do one of either or even both:

1) Push a whole user list from a master server to all ESP-RFID devices via some kind of a POST request
-or-
2) Have the ESP-RFID device continuously interrogate a URL with a get request to update its list of users and their associated weigand codes

(1) and (2) are not mutually exclusive, they could work great together in concert.

windy54, 2022-10-09:
Not sure about 2, I would sooner “manually” update a list of users.  When I have been testing the dev branch with a view to updating our software, I have found the MQTT approach to be reliable as long as I add a delay between transmission of MQTT messages. ( I have not looked at using the response from the esp-rfid) we have about 100 users and I have found when restoring from an old file to be unreliable. About 20 members get updates, still trying to find out what is going on.

danielschonfeld, 2022-10-09:
(1) works, but really anything that is a bit more atomic than one at a time.  When controlling a few of these devices over several locations, with one centralized server (imagine multiple buildings in the same community) - it's more reassuring to know, that once you've updated the centralized server a full fresh copy has been sent to the associated controller and has been accepted.  Rather than figuring out the deltas and sending them one at a time, with each one, increasing the chance of someone being missed.

matjack1, 2022-10-22:
@danielschonfeld makes sense. How many users are you looking to import in one go?

I'm no MQTT expert, but would that be possible to send a big list of users in one single MQTT message? Would that be reliable?