omersiar/esp-rfid #539: Relay2 to mqtt

henfri, 2022-10-25:
Hello,

I see that it is possible to activate more than one relay. But unfortunately, this does not seem to work on the mqtt side. 
Currently, when presenting a tag of a user for which Access1 is disabled and Access2 is enabled, I only get

`esp-rfid/tag {"cmd":"log","uid":"3db","username":"henfri","access":"Disabled","time":1666716153}
`
of course, I could now react on the uid or so, but that would mean that all the user-management in esp-rfid is kind of disabled.

Is it possible to get a message
`esp-rfid/io/lock2 UNLOCKED` 
or similar, when access2 is presented.

Greetings,
Hendrik

matjack1, 2022-10-28:
hey @henfri I know about this :( I have a branch in the working: https://github.com/esprfid/esp-rfid/compare/dev...doorname where we add a name for each relay, and send all the open relays out via MQTT. This is going to be part of v2, so hopefully coming in the near future :)

matjack1, 2022-11-06:
Hey @henfri I was wondering about the use case of having one relay to open and another one not. What's your use case for that?

@omersiar do you have references to why this was introduced?

I mean, if you have multiple doors to open to get somewhere passing an RFID tag, what will happen if you don't have a certain permission? That you are trapped somewhere? Wouldn't be more sensible to have multiple esp-rfid systems instead?

henfri, 2022-11-06:
Hi @matjack1 ,

:-) That's easy:
Left hand Fingerprint open Door of House
Right hand Fingerprint open Door of Garage

matjack1, 2022-11-06:
Thank you @henfri!

So you have one sensor in that controls two relays in distant places? Wouldn't make more sense to have one esp-rfid for each door?

Bear in mind that the long wires that you need to connect to the esp-rfid have non-encrypted traffic, easy to read/write. What do you think?

henfri, 2022-11-10:
Hello,

I also created an issue on this.
To make it very simple, I could imagine mis-using the Names of the fingerprints, e.g.

Paul Left Thumb (GPIO3)
Paul Right Thumb (GPIO4)
Marry Right Thumb
This would trigger GPIO3 if Paul uses his left thumb, GPIO4 on his Right Thumb and for Marry, it would only inform via mqtt.

Best regards,
Hendrik

matjack1, 2023-10-06:
Hey @henfri I've done what I could here, now when you have more doors you get via MQTT the permission for each door.

There is one limitation though, access on the first door is mandatory, the other ones are optional.

So in your case you could have a fake first door and use the remaining three to configure your doors?

Anyway, the code in in the `dev` branch. If you are still interested please try that out and let me know :) I'm closing this for now, but feel free to reopen if necessary.

closed 2023-10-06 by matjack1