omersiar/esp-rfid #542: Reader HID multiclass SE RP10

stupsi099, 2022-10-30:
with forV2Board Date 29.11.2019 the reader works as Wiegand - only webinterface is not stayable.
with update to newer build forV2Board or generic.bin i cant get the reader to work. Same Pinout same config - it wont work.
Reader Type= Wiegand
Pin D0= GPIO-5  green cabel from reader
Pin D1= GPIO-4  white cable from reader
Pin code request= false
WiFi LED= GPIO-2
Door Status Pin= GPIO-16 official board
if i change back to old version it works again - with newer one it wont read any badge.

any idea why ??
(testet on 3 official green boards)
br rupert

matjack1, 2022-11-06:
Hey @stupsi099 could you please try using the latest dev build instead? You can find it here: https://github.com/esprfid/esp-rfid/actions/runs/3404792413 at the bottom of the page.

Let me know if that helps! Thanks.

stupsi099, 2022-11-06:
did not help - not work with the new version.
i set wlan to the internal led and can see the esp8266 is rebooting if you go to the log tab and switch between access and log tab...
-> i see fetching data -> reboot -> in log i see reboot!

14 | INFO | sys | System setup completed, running |   | 1
13 | INFO | wifi | WiFi is connected | WLanZ1 192.168.1.114 | 1
12 | INFO | door | Door Closed |   | 2
11 | INFO | sys | System setup completed, running |   | 2
10 | INFO | wifi | WiFi is connected | WLanZ1 192.168.1.114 | 2
9 | INFO | websrv | Login success! | 192.168.1.101 | 2022-11-06-18:59:23
8 | INFO | door | Door Closed |   | 11
7 | INFO | sys | System setup completed, running |   | 11
6 | INFO | wifi | WiFi is connected | WLanZ1 192.168.1.114 | 11
5 | INFO | sys | System is going to reboot |   | 2022-11-06-18:59:06

br rupert

matjack1, 2022-11-06:
thank you for the feedback, I'm going to check

stupsi099, 2022-11-11:
i try this build "Add pin code only option to open door #71 platformio
Option: RFID Card without asking Pin code

not working!! (users -> add card -> nothing, did not read)

br rupert

stupsi099, 2023-01-23:
with platformio #104 same problem.
card wont read  - my config if i did anything wrong with original green board.

  },
  "hardware": {
    "readertype": 1,
    "wgd0pin": 5,                                   green cable
    "wgd1pin": 4,                                   white cable
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 2,
    "rtype": 1,
    "ltype": 0,
    "rpin": 13,
    "rtime": 400,
    "beeperpin": 255,
    "ledwaitingpin": 255,
    "openlockpin": 255,
    "doorbellpin": 255,
    "accessdeniedpin": 255,
    "useridstoragemode": "decimal",
    "requirepincodeafterrfid": false,
    "allowpincodeonly": false,
    "doorstatpin": 255,
    "maxOpenDoorTime": 0,
    "numrelays": 1,
    "relay2": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay3": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay4": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    }
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "-----",
    "openinghours": [
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111"
    ]

matjack1, 2023-01-23:
Hey @stupsi099, just checking, are you referring to this build here: https://github.com/esprfid/esp-rfid/actions/runs/3980992874 ?

Anyway thank you for your settings, I'm going to check soon

stupsi099, 2023-01-23:
yes i use the last one #105 from pfatformio

 

with the old one from the release (29.Nov.2019) it works with this pinout and reader.

 

newer did not scan the card. the reader read but no output on the "user tab" to add a new user

 

br rupert

 

Gesendet: Montag, 23. Januar 2023 um 20:29 Uhr

matjack1, 2023-01-23:
Sorry, one more question, when you say that the reader reads, do you see anything on the serial logs? Or the reading is just a "beep" that you hear?

I don't have an idea on what's wrong on the new build and unfortunately I don't have the reader that you have to test :(

If you have a spare one to ship I'm more than happy to fix it for you, if you don't have any better option!

stupsi099, 2023-01-23:
Beep from reader. But its internal - i have spare reader! You can have one

matjack1, 2023-01-23:
Cool, feel free to send me an email to my public email (on Github profile) and I'll let you know my address. Thanks!

matjack1, 2023-02-14:
Hey @stupsi099 have a go with this: https://github.com/esprfid/esp-rfid/pull/582

Thank you very much for sending over the reader, I've learnt new stuff around Wiegand readers :)

stupsi099, 2023-02-15:
now the reader works with all cards i have 125khz, 13.xMhz Desfire and old one.
but there are strange ID`s when i read the badge/fob

old version:   1918649604         new software: 959324802             Desfire EV2 8k        Serial 04:45:5C:F2:E4:5E:80
if i switch to hex
new software: 3b3e7782    Mifare Classic 1k   Serial 04:EF:7C:76

back to dec
new software: PICC Type 32    1816220      125kHz Fob       Serial 0003632440   (-50%)      
old version: PICC Type 20      3632440                  same as on the chip 
(every card on old version are PICC Type 20 - on new version are PICC Type 32)
-> but all cards readed by new software!!! -> problem is only wrong ID. ->by 125kHz Chip its 50% from the number :-)
hope it helps
br rupert

stupsi099, 2023-03-28:
new version works now same as old version. All cards accepted and readed.
thx for your work
br Rupert

matjack1, 2023-03-28:
Thank you very much for your help! I've learned something new and important :)