omersiar/esp-rfid #546: Fixing broken mfrc522 and rdm6300 scanners on the latest dev firmware

Aleks130699, 2022-11-07:
Today I devoted all my free time to this project in search of problems in which my readers did not work, I fully studied the part of the code responsible for rfid, I was finally able to make everything work, the first problem due to which the readers did not show signs of life at all, as already written in other discussions, was the line rfidState = cardSwiped;, which was absent in the processing of the mfrc522, pn532 and rdm6300 readers, after adding this line, the readers started working, but the cards caused the module to reboot, after understanding in detail it turned out that the keys with the Admin label were handled perfectly, and also unfamiliar keys. All keys marked Always, immediately reloaded the module. Going further, I tried to comment out these lines
`#endif

	if (accountType == ACCESS_GRANTED)
	{
		// unsigned long validL = json["validuntil"];
		// unsigned long nowL = now();
		// int hourTz = hour() + config.timeZone;
		// bool officeHour = config.openingHours[weekdayFromMonday(weekday())][hourTz] == '1';
		// if (validL < nowL)
		// {
		// 	processingState = expired;
		// } else if (!officeHour)
		// {
		// 	processingState = notValid;
		// } else
		// {
			processingState = valid;
		// }
	} else if (accountType == ACCESS_ADMIN)
	{
		doEnableWifi = true;
		processingState = validAdmin;
	} else {
		processingState = notValid;
	}

`
And this fixed the whole problem of restarting the module when reading cards.
But also studying the code, I did not see the reverse change of the rfidState parameter, which I think should happen with the line rfidState = waitingCard;.

Aleks130699, 2022-11-07:
#506

matjack1, 2022-11-07:
Thank you for the details. I'm going to check this class of hardware soon. Surely before finalising the v2.

Aleks130699, 2022-11-07:
@matjack1 i think this should be a higher priority fix, since without it, use with these modules is not possible

matjack1, 2022-11-07:
I've done the easy part here: https://github.com/esprfid/esp-rfid/pull/547

But I'm not sure about the bug later on. This pulls the information from the user, so maybe there's a problem in your user database? Can you please double check if the user has the necessary data?

Aleks130699, 2022-11-07:
I completely deleted the entire database and wrote down new keys, and when it is set to Always, a reboot occurs, I checked all the necessary data.

Aleks130699, 2022-11-07:
You may have to record with a complete overwrite, so that there is definitely no data left from the old firmware

matjack1, 2022-11-07:
Can you please share your settings? You can download them in the "Backup & Restore" section. I guess I have some different settngs than you.

Aleks130699, 2022-11-07:
How can I send you files?

matjack1, 2022-11-07:
ok, looks like you don't have the opening hours settings. Can you please go in the general settings page and check if that's selected?

In any case, this is a bug as we should support also previous configurations, as yours, without breaking. If you want to try also with the code in the PR here: #547 it might work! :)

Aleks130699, 2022-11-07:
Yes, the settings page has all the checkmarks on the time, Okay, I'll try tomorrow with the changes, and with a full reset.

matjack1, 2023-01-23:
@Aleks130699 I'm closing this for now as I didn't get any feedback, also here: https://github.com/esprfid/esp-rfid/pull/547

But if you still have problems let me know!

PR #547 (2022-12-04): Fix support for readers other than Wiegand
Fixes #546
closed 2023-01-23 by matjack1