omersiar/esp-rfid #548: WLAN AP is not working

Bertram-561, 2022-11-09:
I bought the esp-rfid Blue Board and powered it on and no Access Point to connect showed up. I flashed the firmware once again , still no AP. Is the board broken  or am I missing something ?
Regards 
Bertram

matjack1, 2022-11-10:
hey @Bertram-561 have you tried flashing a dev version? You can try the last one from here: https://github.com/esprfid/esp-rfid/actions/runs/3404792413

@nardev do you confirm that the latest dev build works on your blue board?

majuss, 2022-12-14:
I can confirm the issue. Used every bin available also the dev snapshot from the pipeline. Flashed with:
`esptool.py --port /dev/ttyUSB0 write_flash 0x1000 generic.bin`
No hotspot is opening, also tried the debug build, I can't see any serial output on `screen /dev/ttyUSB0 9600`.

matjack1, 2022-12-14:
@majuss if you don't see anything on the serial with the debug build, seems to me like a hardware problem :(

nardev, 2022-12-14:
Well, as i said, it could be that the EPS doesn't boot at all.. But than the programming would also be a problematic...

nardev, 2022-12-14:
Any other arduino sketch works fine?

majuss, 2022-12-14:
Yes ESPHome for example works. Blue LED is lit up constanly.

majuss, 2022-12-14:
Update on my side. I flashed with address 0x00000 like in the flash.bat and now it worked fine! Maybe include a flashing example in the README.

Is there a way to grab other wiegand events like doorbell, or pincode entries on a keypad? I have this keypad: https://de.aliexpress.com/item/1005004210605222.html

matjack1, 2022-12-14:
Hey @majuss thank you for the feedback!

I'm just using Platformio so I never flashed like you did. Would be nice to have a script like the flash.bat for Linux/Mac, if you want to give it a go I'm surely going to review it, thank you!

closed 2026-10-01 by matjack1