omersiar/esp-rfid #556: Incorrect UID read with Wiegand 32 bit device - anyone know protocol

windy54, 2022-11-18:
We currently use a 522 reader and have been donated several Wiegand devices.

These output 32 bits when a card is read.

The displayed UID is totally different to the UID stored on the card.

I have investigated the code and have a fix but this is to the Wiegand.cpp library file not the esp-rfid code.

If we had not already been using a 522 reader this issue would never have arisen.

So my question is, does anyone have a reference to the Wiegand 32 bit protocol? I can't find one.

I am going to post a similar question in the github repository for the code.

evazzoler, 2022-11-18:
Try to check if D0 and D1 are inverted.

windy54, 2022-11-18:
They are okay, I originally had them inverted on my logic analyser. When I work through manually the errors occur when the last bit is read in.

matjack1, 2023-01-23:
hey @windy54 do you confirm this is something to do with the Wiegand library? Am I right?

I'm going to close this for now, but if you still need help on this one let me know. Maybe can you provide a couple of examples of RFID reads from the two sensors for the same card?

closed 2023-01-23 by matjack1

windy54, 2023-01-23:
Hi,Yes I have edited the wiegand.cpp file, I will send the edits to you.CheersSteve Gale