omersiar/esp-rfid #567: bootloop when setup WPA pass with ~~

p14c3k, 2022-12-16:
Software going on bootloop on wemos D1 mini when WPA pass including ~~ during initiall WLAN  setup.

For return example:  Xxxx~~Yyyyyy

To have it back working, need to flash erase bin

matjack1, 2022-12-16:
Thank you for the report!

Can you please confirm that you are testing the dev branch?

matjack1, 2023-01-23:
I've tried reproducing with a password containing ~~ and it works for me.

Maybe the cause of your issue was somewhere else?

Please, try again with the binary that you can find here: https://github.com/esprfid/esp-rfid/actions/runs/3980992874 it's the last build.

I'm closing for now, but if you still encounter a problem, please reopen and also share your configuration, that you can download from the backup&restore section in the web interface of your esp-rfid. Thanks!

closed 2023-01-23 by matjack1