omersiar/esp-rfid #568: Cant save reader settings in new build

stupsi099, 2022-12-19:
Platformio build #100
green official board and wiegand reader HID RP10

If you try to change wlan and press save yellow line apears and it works
if you try to change reader settings and press save - nothing - no yellow line

konradhuebner, 2022-12-26:
I tried the same today, also does not work for me. It does not show the "save" button, and also does not restore the settings in the webpage. It does however save parts of it. The Wiegand setting incl. D0/D1 were saved, but the beeper and LED were not

hinkma, 2022-12-29:
I try setup Wiegand. Save button web page not do nothing to log. 
Second one problem is that 'LED access denied OFF' running all the time  terminal.

hinkma, 2022-12-29:
I make Factory reset and now LED access denied OFF' running not active any more.
Every time when I click save button, ESP reboot itself.

stupsi099, 2023-01-10:
With build #101 save works again.

matjack1, 2023-01-12:
Thank you for the feedback. I'm going to close for now, but feel free to reopen if necessary

closed 2023-01-12 by matjack1