omersiar/esp-rfid #574: config.pinCoderequested set true when a 522 reader used in latest dev build?

windy54, 2023-01-19:
I downloaded the dev build last week (Jan 9th) to investigate a couple of issues.

With a 522 reader the cards were not being read and I tracked the issue to this test
`void rfidProcess()
{
	if (rfidState == waitingRfid ||
			(config.pinCodeRequested && rfidState == cardSwiped))
	{
		return;
	}`

it kept returning because config.pinCodeRequested  was true.

The option to change this is only available on the web page when using a wiegand reader.

I cant see where in config.esp it is set true when a 522 reader is used, but have solved the issue by editing config.esp and adding the following
`else if (config.readertype == READER_MFRC522 || config.readertype == READER_MFRC522_RDM6300)
	{
		rfidss = 15;
               **config.pinCodeRequested = false;**
		if (hardware.containsKey("sspin"))
		{
			rfidss = hardware["sspin"];
		}
		int rfidgain = hardware["rfidgain"];
		setupMFRC522Reader(rfidss, rfidgain);
	}`

with the above mod cards are read by my 522 reader

Steve

matjack1, 2023-01-19:
Hey Steve, you are right, we need to set the pincode requested boolean to false if you use another reader different from the wiegand.

Am I right thinking that the wiegand is the only reader with pinpad?

windy54, 2023-01-19:
I am not sure bout the keypad, but the readers we have not been donated at the hackspace with keypad are all wiegand.SteveSent from my iPhone

bumaas, 2023-01-22:
I helped myself by setting the initialization of 'pinCodeRequested' to false in config.h:
`bool pinCodeRequested = false;`

matjack1, 2023-01-23:
@windy54 @bumaas you are right!

Please have a look at this PR: https://github.com/esprfid/esp-rfid/pull/578 if you can test it.

I don't have a PN522 set up at the moment, so I cannot really test it. Thank you!

matjack1, 2023-03-08:
I have merged the fix in the `dev` branch and I'm closing this, please reopen if necessary

closed 2023-03-08 by matjack1