omersiar/esp-rfid #575: Keypad login

timknowlden, 2023-01-20:
I have no set up my keypad weigand reader, am I mistaken but is only the rfid tag supported method of access or can the number pad be used also?

timknowlden, 2023-01-20:
Numbers are picked up in the  add user page, but only the 1st number. So if i want to add PIN 1234 as soon as i press '1' it opens add user.

timknowlden, 2023-01-20:
Forgive me, upload of newer firmware must have failed, I have re-uploaded 1.3.7 (which is listed as 1.3.1) and the keypad now waits for # to be pressed which allows me to enter longer pin numbers.

timknowlden, 2023-01-20:
I don't know where to post this but how can I find, download and install the latest dev branch?

closed 2023-01-20 by timknowlden

matjack1, 2023-01-21:
Hey @timknowlden you can get here the last dev build: https://github.com/esprfid/esp-rfid/suites/10269592798/artifacts/504177320

In general here: https://github.com/esprfid/esp-rfid/actions you can find all the builds for all the branches and commits, OK?

Let me know if you still have problems. Cheers

timknowlden, 2023-01-22:
Thanks, I thought I read somewhere the was home assistant auto discovery and ability to include the bell button from keypad with latest dev build.

matjack1, 2023-01-22:
yes @timknowlden the auto-discovery on Home Assistant is there and should be working :)

For the doorbell, it's part of the HA integration, but I'm not sure if you are talking about something particular on your keypad.

You can drive the doorbell with a pin, I hope this helps.

timknowlden, 2023-01-22:
Sadly I have no option to enable a doorbell pin in the settings, unless I am mistaken and looking in the wrong place?

[image]

matjack1, 2023-01-22:
hey @timknowlden looks like you are not on the `dev` branch, look how it should look like on `dev`:

[image]

timknowlden, 2023-01-22:
Ah that's good to know, perhaps it is not uploading then when I try here to upload latest file:
[image]

matjack1, 2023-01-22:
strange @timknowlden it should work from there too. I'll try myself soon

matjack1, 2023-01-23:
It takes a while to update, if you can try again in debug mode with the serial monitor attached, and if you can share the logs it would help to see what's going on in your setup.

Let me know if you have more news. Maybe you can try again and just wait a bit to make sure that the process has finished.

timknowlden, 2023-01-24:
Yes, I'll try to update again by UI. 

I've also picked up a 12v relay to try and get the doorbell sensor working as my keypad has 2 doorbell wires but needs a 12v supply, the esp only gives 3 or 5v.