omersiar/esp-rfid #584: MQTT restart

timknowlden, 2023-03-10:
Hi, 
Sometimes my device becomes unresponsive via mqtt command. I haven't yet worked out if its due to the device or the server.

I have to restart the esp device to get it to respond again. Is there a way to restart the connection without restarting device and automatically too? 

Running the latest dev build.

Also when I restart my device it sends a message by mqtt to say the door has been opened by the push button. Is there a way to disable this message being sent, or change the message to say device has restarted. 
[image]
[image]
[image]

matjack1, 2023-03-10:
hello @timknowlden are you only sending MQTT commands and then it stops responding or are you also using the Web UI?

If MQTT only, which messages are you sending?

timknowlden, 2023-03-10:
It's only mqtt commands that it is not working with.

The Web ui is still functional and that is how I'm restarting the device

matjack1, 2023-03-10:
Can you share a log of calls that you are sending/receiving from the esp-rfid, so that I can try reproducing? Or some steps that you take that reproduce the problem. I have a very stable system with MQTT (mostly sending from esp-rfid)

timknowlden, 2023-03-11:
I am using home assistant to issue commands via mosquito broker, the log will be flooded with messages to and from other devices. It can last for days on end before becoming unresponsive. 

If the message on restart was not that the door was operated then I could set the device to restart daily?

matjack1, 2023-03-11:
Yes @timknowlden that's what I would do! You can setup the autorestart from here:

[image]

You can also set a custom time, if 24h are too much.

I'm going to close this for now, but reopen if this is not good enough. Unfortunately there's only so much we can do from esp-rfid to mitigate these type of issues.

closed 2023-03-11 by matjack1