omersiar/esp-rfid #594: Door Status ?

Tiefflieger1, 2023-04-27:
Hello
I have the problem when I activate the door status via Gpio-16, the system no longer works properly. doesn't let in chips. can someone help me? I've also upgraded the new version. unfortunately brings nothing

matjack1, 2023-04-27:
hey @Tiefflieger1 unfortunately the GPIO usage depends on your actual device and you need to try different GPIO combinations to find out the one that works for you :(

You can maybe start from here: https://github.com/esprfid/esp-rfid/tree/dev#pin-layout if you find configurations that don't work for you let me know.

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1