omersiar/esp-rfid #599: Version >118 not work with HID Reader

stupsi099, 2023-08-26:
newer version not work with this reader - after update to Build 130 ESP reboots only
with Build 124 it works but no card was read

where to find version 118 for download ??
br Rupert

matjack1, 2023-08-26:
Hey @stupsi099 can you please confirm that the latest dev build, that you can find here: https://github.com/esprfid/esp-rfid/suites/15398950135/artifacts/879281209 is broken for you?

If so, can you please share your configuration file so that I can try to replicate on my end? Is the reader the one that you have shared with me previously?

stupsi099, 2023-08-29:
i update to Build 125 and use my old config and the system was working again, only timesync is not working. I need to buy 2 more green board so i have a testsystem again. My 3 System are working and online.... and in use.
on tindy the boards are sold out so i have to wait.
br Rupert

stupsi099, 2023-09-20:
update to Build 131 with the new board . make full reset, try to config and save 
system comes up and read card   ------ BUT -------
if you look inside the running system every settings are default.
blue buard needs ID13 and ID12 -> settings saved and working but if i look after full restart into the running system all parameter back to default  - ID5 ID4

matjack1, 2023-09-29:
@stupsi099 this issue is a duplicate of https://github.com/esprfid/esp-rfid/issues/602 ?

Looks like the problem is with not being able to correctly save the configuration, right?

stupsi099, 2023-10-06:
with the "dev" 133,134 Build the problem is gone!

stupsi99

closed 2023-10-06 by matjack1