omersiar/esp-rfid #602: Problem writing the configuration

geraldFS, 2023-09-17:
Hello,

I have a problem when I install the current image, there are problems with writing the Conf file.
Values are not entered or when changed, everything is reset to the first setting.

Somehow everything is inadequate, I have already emptied the entire flash memory with blank4mb.bin and reinstalled everything - not satisfactory.

I hope my problem can be reproduced.

I use an esp12F with 2 relays - This hardware has been running for quite some time.

Greetings
Gerald

stupsi099, 2023-09-20:
same problem. looks the parameter were stored in the system but on webinterface always default displayed.
if you try to config and save - all parameter back to default like displayed, and the system is back to ap-mode..

matjack1, 2023-09-20:
Thank you @geraldFS and @stupsi099 I have the same problem, I think.

It's very strange as sometimes it works and sometimes everything is gone. I was thinking that it was a hardware problem, but if you have the same it must be a software one. I'll check and report on this issue.

stupsi099, 2023-09-27:
with build_124 save works, but no ntp sync!!

matjack1, 2023-09-29:
@stupsi099 I've checked build 124, but it's very close to the current one, I cannot see any difference that could impact configuration.

Is it not working all the time or only sometimes? I get that normally it works, but sometimes it breaks. It's very difficult to debug :(

stupsi099, 2023-10-02:
with dev build_133 config works (after update from web i have to flash with serial again! - esp only blinks)
config works again, only problem with ntp not update my system... and logfiles crashing the system.
If you start logfile view only fetching data and then system reboot..

stupsi099, 2023-10-06:
Build dev_134 
config works - (config, save, restart -> all OK) 
ntp works - (on startup or save with reboot - ntp works soon, if system restart after crash - ntp only after the default set time)
system crash - (never on running system, often on web with the logs output -> Event log)

is it possible to view the systemtime on the startup screen ??
and to view only the last 20-30 entrys in the log files (all other only with download button) to increase the stability ?

stupsi

matjack1, 2023-10-06:
hey @stupsi099 thank you very much for your feedback.

What you are asking about event logs makes a lot of sense. The overall problem is with websockets. After a while they corrupt memory and esp-rfid crashes. The more features we have added the worst it is as the available memory is less and less.

So, I've tried to minimise the problem by getting only one page at the time rather than the full list here: https://github.com/esprfid/esp-rfid/pull/577 but I'm not sure it's working better. That PR is only about the users table. If that works better for you I can try to apply the same approach also to the events table. Let me know what you think. Thank you!

stupsi099, 2023-10-06:
update to dev_137 -> config work, only reboot with log view.
down to fixwebsocket_136 -> config in memory after reboot, but in web all default 
update to dev_137 -> config work again

matjack1, 2023-10-13:
yes, with fixwebsockets I've found the problem. I'm not sure if I'm going ahead with that PR actually. Maybe differently from what I've done as I've found that we lose some feature by changing the pagination (like you cannot filter by user anymore).

Use build from the dev branch so we should start from a fairly stable state :)

I'm closing for now, if you still have problems let me know.

closed 2023-10-13 by matjack1