omersiar/esp-rfid #607: Build 163  - Uptime in Status only a number...

stupsi099, 2023-11-15:
if you look in status field you can see only a rising number (no date/time)
not shure if its a bug or a feature.

br
rupert

matjack1, 2023-11-17:
Hey @stupsi099 the uptime is in seconds now. I agree that it would be nicer if it was parsed and shown in days/hours/seconds.

If anyone wants to try improving that let me know, I can help giving you directions on how to implement this.

pvtex, 2024-06-08:
this is no problem just change the file _wsRepsonse.esp_
in the function 
```
void ICACHE_FLASH_ATTR sendStatus(AsyncWebSocketClient *client)
```

change this line:
```
root["uptime"] = uptimeSeconds;
```
tp:
```
int h = uptimeSeconds / 3600;
  	int rem = uptimeSeconds % 3600;
  	int m = rem / 60;
  	int s = rem % 60;
	char uptimebuffer[9];
	sprintf(uptimebuffer, "%02d:%02d:%02d", h, m, s);

	root["uptime"] = uptimebuffer;
```

omersiar, 2024-06-12:
Rather than making the transformation on limited esp hardware, it would be better to handle it on Client side (browser).

matjack1, 2024-09-21:
True, but I think that's also fine as pvtex did. It's now on dev, closing this!

closed 2024-09-21 by matjack1