omersiar/esp-rfid #608: stability problem after update to Build-163 and solution

stupsi099, 2023-11-20:
i update 5 Systems to generic build 163 and got same stability problems after update
(official 3x green Board, 2x blue Board)

if you log in it takes 5 second to update the status screen, if you go into any settings you will push back to status screen after a few second.

**Solution:**
update to build 163
factory reset ->  you will get an AP with esp-rfidXXXXX     (->XXX unique number<-)
login with 192.168.4.1 pw:admin and update again to build 163
factory reset again -> you will get an AP with esp-rfid  (->without number<-)
config manual (do not use your old config file!!!)
power off and restart - now my system works

if you import your user file - you have to edit and clear the pincode field again.
[image]
[image]
because sometimes there are wrong data inside.

if i login to the reader the status info appear immediately.
not shure if you can skip some task above

br rupert

matjack1, 2023-11-21:
hello @stupsi099 thank you very much for testing the new version!

Can you please share your configuration file when it has problems? So that I can try reproducing your initial situation and see what's going on. Since you say that if you reset and start from scratch it works, I think it's a problem with your specific configuration that I have not tested yet.

If you prefer to share it via mail it's fine too.

pvtex, 2024-06-08:
pincode problematic should be fixed as in #641

matjack1, 2024-06-11:
This should be now fixed by #645 

Thank you @pvtex!

@stupsi099 if you want to try again a new build from `dev`, this should be fixed. I'm closing for now but if it's still there let me know.

closed 2024-06-11 by matjack1