omersiar/esp-rfid #609: Relais turns ON while restart

CNC-Maniac, 2023-11-25:
Hello, I have now updated from v1.2 to v2RC. 
Everything looks good so far and MQTT has become much more convenient. 
But since I had stability problems with Wifi, I set "auto restart" to 7200sec under general settings. 
Every time the controller restarts, the relay switches on briefly and my front door is no longer locked.

matjack1, 2023-11-27:
That's not good!

If you can share the configuration file I can try replicating your setup and issuing a fix.

If you don't use the web UI, the MQTT-side alone should be fairly stable and you shouldn't need a recurring reset.

Thank you for the feedback!

matjack1, 2023-11-28:
Thank you very much @Divyastra ! That's a good point.

@CNC-Maniac when you can share your config file let me know, I'll have a go :)

CNC-Maniac, 2023-11-29:
Sorry for the delay. Here is my config (I am using the original hardware):
`{
  "command": "configfile",
  "network": {
    "bssid": "DC:39:6F:0D:18:6D",
    "ssid": "XXXXXXXXX",
    "wmode": 0,
    "hide": 0,
    "pswd": "XXXXXXXXX",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readerType": 1,
    "wgd0pin": 5,
    "wgd1pin": 4,
    "sspin": 15,
    "rfidgain": 32,
    "wifipin": 2,
    "rtype": 1,
    "rpin": 13,
    "rtime": 1500,
    "ltype": 0,
    "buttonpin": 255,
    "doorstatpin": 255,
    "maxOpenDoorTime": 0,
    "doorbellpin": 255,
    "accessdeniedpin": 255,
    "openlockpin": 255,
    "beeperpin": 255,
    "ledwaitingpin": 255,
    "useridstoragemode": "decimal",
    "removeparitybits": true,
    "numrelays": 1,
    "relay2": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay3": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay4": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "readertype": 1,
    "requirepincodeafterrfid": false,
    "allowpincodeonly": false,
    "doorname": "Garage"
  },
  "general": {
    "hostnm": "esp-rfid-Garage",
    "restart": 0,
    "pswd": "Anke1974!",
    "openinghours": [
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111"
    ]
  },
  "mqtt": {
    "enabled": 1,
    "host": "192.168.178.246",
    "port": 1883,
    "topic": "rfid/Garage",
    "user": "",
    "pswd": "",
    "syncrate": "60",
    "mqttlog": 1,
    "autotopic": true,
    "mqttha": 1
  },
  "ntp": {
    "server": "fritz.box",
    "interval": 30,
    "timezone": 1,
    "tzinfo": "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00"
  }
**}`**
I tooked the autoreboot off right now, but anyway this is also a problem after a power failure

matjack1, 2023-12-19:
Hello @CNC-Maniac I've finally manged to try your configuration and with my ESP the relay is not triggered on reboot. I'm using GPIO-13 on a NodeMCU v3 (an ESP8266 dev board).

If anyone else can reproduce please let me know, meanwhile try using other GPIO pins as Divyastra suggested.

I'm closing for now, but please reopen if the problem is with the software.

closed 2023-12-19 by matjack1