omersiar/esp-rfid #611: esp-rfid keeps disconnecting from mqtt

IVCrossway, 2023-12-09:
esp-rfid latest dev branch. It keeps disconnecting from mqtt and this results in home assistant thinking the door is open.
Rebooting the esp solves the problem temporarly. The disconnects seem to be random and other mqtt devices are not experiencing this issue.

Mosquitto logs show:

```
1702112729: New client connected from 192.168.188.148:60303 as esp8266-f9c4d6 (p2, c1, k15, u'mqtthome').
1702113162: Client esp8266-f9c4d6 has exceeded timeout, disconnecting.
1702113237: New connection from 192.168.188.148:62236 on port 1883.
1702113237: New client connected from 192.168.188.148:62236 as esp8266-f9c4d6 (p2, c1, k15, u'mqtthome').
1702113692: Saving in-memory database to /mosquitto/data//mosquitto.db.
1702114968: Client esp8266-f9c4d6 has exceeded timeout, disconnecting.
1702115035: New connection from 192.168.188.148:63326 on port 1883.
1702115035: New client connected from 192.168.188.148:63326 as esp8266-f9c4d6 (p2, c1, k15, u'mqtthome').
1702115493: Saving in-memory database to /mosquitto/data//mosquitto.db.
1702116054: Client esp8266-f9c4d6 has exceeded timeout, disconnecting.
```

While esp-rfid just shows:
[image]

matjack1, 2023-12-19:
hey @IVCrossway I'm sorry for your problem :(

Unfortunately this is not something that happens to me, MQTT connection is reliable to me.

It's strange that HA thinks that the door is open when esp-rfid is disconnected, is it something we can fix on esp-rfid side?

Instead on the MQTT connection, maybe you have already tried, but is the esp close to the WiFi AP? Also, have you tried another board? Sometimes boards are not very reliable, maybe try another one. Finally, is the esp-rfid reconnecting to MQTT automatically or only if you reboot?

IVCrossway, 2023-12-19:
> It's strange that HA thinks that the door is open when esp-rfid is disconnected, is it something we can fix on esp-rfid side?
I think this should be handled by home assistant itself, I just have to figure out how.

> Instead on the MQTT connection, maybe you have already tried, but is the esp close to the WiFi AP? Also, have you tried another board? Sometimes boards are not very reliable, maybe try another one. Finally, is the esp-rfid reconnecting to MQTT automatically or only if you reboot?

The wifi connection I guess is fine. The esp is connected to a Range Extender to get the best possible signal quality. I don't know if it's dropping packets or something but, correct me if I'm wrong, this should not be an isseue since packet drops happen every so often and the packet will be delivered before the mqtt timeout. (I'll further investigate into my network reliability)

The fact is that after some time (could be hours, could be minutes... It's random like the disconnects) it reconnects itself without any reboot. And in this timeframe the web interface is working just fine, so I guess this is not a connectivity issue.

matjack1, 2024-06-11:
Hey @IVCrossway are you still interested and using this project? Do you have any news about this issue? Since I've not received any similar issue and not experienced it myself I'm inclined to close this. Let me know how it's going, thanks!

IVCrossway, 2024-06-12:
Yes I'm still using it. As a "workaround" I added another esp with esp-home to have an app control of the garage door. I needed the MQTT connection just to open the door using my phone.
If you want you can close the issue, unfortunatly I do not have time to troubleshoot this any further.

matjack1, 2024-07-16:
I'm leaving this here, in case similar reports come through, but right now I don't have clues on what to work on, sorry :(

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1