omersiar/esp-rfid #615: Release v2.0.0.rc0 doesn't work properly

MatteoSalviTech, 2024-01-13:
Hi! I've tried this solution on October 28 and I was so enthusiast since it worked perfectly. This new v2.0.0.rc0 doesn't work properly... can you please send me the generic.bin file before v2.0.0.rc0?
Here are some of the problems I've discovered:
- Not working in AP mode
- Page loading amount of time is very long and sometimes pages such as settings or users doesn't load at all
- User page doesn't load properly
- Cannot use PIN only mode with my wiegand reader (it read one pressed key at a time)
- Problems persists if I connect it to a Wi-Fi network in client mode

matjack1, 2024-01-16:
Hey @MatteoSalviTech ciao! :)

Please have a go with the binaries from here again: https://github.com/esprfid/esp-rfid/releases/tag/2.0.0.rc0 I've just reuploaded them with the latest bin file I have built and tested on my device. I cannot experience the problems you have, so if the problems persist, please send also a dump of the configuration to be sure I can reproduce your issue. Thank you for testing the new version!

MatteoSalviTech, 2024-01-17:
Ciao @matjack1 ! So, what I've found out is that the ESP is so slow in AP fallback mode. If i connect it in client mode pages reload at an acceptable speed.
I flashed my ESP8266MOD 12-F with the latest debug.bin file after I erased the flash memory with the blank4mb.bin file.

My actual configuration is the following:

`{
  "command": "configfile",
  "network": {
    "bssid": "E6:63:DA:15:BE:02",
    "ssid": "IoT @ SalviCalzolariNetwork",
    "wmode": 0,
    "hide": 0,
    "pswd": "iOtcasasalvi1312xm",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0",
    "fallbackmode": "1"
  },
  "hardware": {
    "readertype": 1,
    "wgd0pin": 12,
    "wgd1pin": 13,
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 2,
    "rtype": 1,
    "ltype": 0,
    "rpin": 14,
    "rtime": 2000,
    "doorname": "Door",
    "beeperpin": 255,
    "ledwaitingpin": 255,
    "openlockpin": 255,
    "doorbellpin": 255,
    "accessdeniedpin": 255,
    "useridstoragemode": "decimal",
    "requirepincodeafterrfid": false,
    "allowpincodeonly": true,
    "removeparitybits": true,
    "doorstatpin": 255,
    "maxOpenDoorTime": 0,
    "relay2": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay3": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay4": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "numrelays": 1
  },
  "general": {
    "hostnm": "esp-rfid",
    "restart": 0,
    "pswd": "admin",
    "openinghours": [
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111"
    ]
  },
  "mqtt": {
    "enabled": 0,
    "host": "",
    "port": 1883,
    "topic": "",
    "autotopic": 0,
    "user": "",
    "pswd": "",
    "syncrate": 180,
    "mqttlog": 0
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "tzinfo": ""
  }
}`

My main problem is now adding a PIN to users. The PIN is recognized as scanned RFID tag and it stops reading when I press the 2 key on my keypad instead of the # key.

In the following log I've scanned an RFID tag (not associated to an authorized user) then I entered "135". When I press the 5 key it interprets it as escape key. Then I entered "132" and "134". When I press 2 or 4 it interprets them as enter key. It also seems that key 4 and key 2 are interpreted as the same wiegand code.

[ INFO ] ESP RFID v2.0.0
Flash real id:   00164054
Flash real size: 4194304

Flash ide  size: 4194304
Flash ide speed: 40000000
Flash ide mode:  DIO
Flash Chip configuration ok.

[ INFO ] Config file found
[ INFO ] Trying to setup RFID Hardware
[ INFO ] Configuration done.
[ INFO ] Trying to connect WiFi: IoT @ SalviCalzolariNetwork
[ INFO ] WiFi BSSID: 230:99:218:21:190:2
[ INFO ] BSSID locked
[ INFO ] WiFi STA Connected
LED WiFi OFF
[ INFO ] WiFi IP Connected
LED WiFi ON
[ INFO ] wifi | WiFi is connected | IoT @ SalviCalzolariNetwork 192.168.10.149
[ INFO ] Trying to setup NTP Server
[ INFO ] sys | System setup completed, running | 
[ INFO ] PICC's UID: 13318882
[ WARN ] rfid | Unknown rfid tag is scanned | 13318882 26
[ INFO ] Read timeout or clean after read
Keycode captured. . .
Keycode captured. . .
Stop capture keycode . . .
[ INFO ] PICC's pin code: 1412
[ WARN ] rfid | Unknown rfid tag is scanned | 1412 
[ INFO ] Read timeout or clean after read
Keycode captured. . .
Keycode captured. . .
Stop capture keycode . . .
[ INFO ] PICC's pin code: 1412
[ WARN ] rfid | Unknown rfid tag is scanned | 1412 
[ INFO ] Read timeout or clean after read

Another problem is that it interprets the entered PIN as a scanned RFID tag, so it never populates the PIN code field.

croggero, 2024-02-21:
I have had the same issue regarding AP mode or if Internet connection is not available.  `bool getNTPtime(int sec)` will continue to delay the main loop for 10 seconds every time. Should this be adjusted to automatically fail and retry within 10 seconds but not block the main loop from executing? This change was implemented in #604

croggero, 2024-03-03:
I have a fork with the changes I made to just poll the NTP time every 10 seconds, if it fails it will continue to add to the last successful sync. This works for AP mode (although the time will be incorrect). The commit  on my fork is [here.](https://github.com/croggero/esp-rfid/commit/ec627f327286cbbd6f2dc2ad8353ffd43bd7e61e)

matjack1, 2024-03-24:
@MatteoSalviTech from what you are saying about pressing 2 for confirming the PIN code instead of #, I think the problem is with your Wiegand configuration. Sometimes it might be because you swapped the D+ and D- or because you don't have a common ground between ESP and Wiegand reader. Let me know!

@croggero thank you for pointing that out, you are right! I have integrated the function straight from the Arduino library, but you are right about pausing for 10 seconds :( Would you be able to open a pull request in this project to fix this? Thanks!

matjack1, 2024-04-09:
I'm closing this as it's now fixed on `dev`, thanks to @croggero :)

PR #626 (2024-04-09): Fix for NTP blocking loop if connection is blocked
Fixes #615 
NTP was blocking the loop causing 10seconds delays when executing when no NTP server could be reached. This fix will check every 10seconds and fall back to the device clock, updating it from the last NTP sync.
closed 2024-04-09 by matjack1