omersiar/esp-rfid #617: Tag UID isn't read while using MFRC522 and v2.0.0.rc0

MaximeSharp, 2024-02-10:
Hi,
While testing the new release, I've noticed that scanning a tag does trigger the device but the tag UID isn't read, so it's impossible to register a new card or trigger the relay pin.

[image]

MQTT does also report that a unknown tag has been scanned and the UID is empty too.

[image]

I've confirmed that my setup is working using the version 1.3.7, and it's working fine.

[image]

ewc-beep, 2024-02-15:
Hi
you can find the solution here -> https://github.com/esprfid/esp-rfid/issues/610

MaximeSharp, 2024-02-16:
Hi,

Thanks, seems like I didn't search enough or used the wrong keywords to find this solution.
I'm out of stock of MFRC522, but once I receive my order I will compile the dev branch to test it again.