omersiar/esp-rfid #618: wemos d1 unstable

plex3r, 2024-02-21:
I bought a bunch of wemos d1 style devices to use however not having much luck.

Running my version of R2 with the previous fixes for RC522 readers but I'm currently trying to use PN532 reader modules as they're smaller.

If I have a PN532 connected the system doesn't appear to broadcast it's wifi SSID

If I go back to a RC522 reader it works but it's slow and card reads are few and far between.

Previously my testing was done with AI-Thinker modules

I guess I'll have to run some diagnostic build to see if there's any errors or what not but I'm just putting this up for now, I'm a little bit frustrated as I need a reliable stable solution for user and machine tracking in the hackspace.

Happy to work on this.

Thanks

plex3r, 2024-02-22:
Something I recalled from one of the other posted issues was about wifi fallback causing delays.

I was testing with it in AP mode but now that I've configured it to connect to my wifi it does seem a lot more stable

Divyastra, 2024-02-22:
@plex3r we've had similar issues as have others, our in service unit uses a nodemcu and we have test platforms on another nodemcu and on the mark 1 ESPRFID board... all give very poor performance in AP mode and in fallback mode either running really slow or causing reboots from the watchdog. Our normal in use machine works tolerably well (though again not without issues) when running in client mode. In order to keep the wifi for the RFID separate from the normal hackspace user wifi we have a RasPi setup as an AP and MQTT broker that the nodemcu connects to in client mode.

I noted a comment yesterday by @croggero in #615 that refers to #604 that might offer clues as to the problems that might be worth investigating
ooo-xxx

matjack1, 2024-03-23:
@plex3r @Divyastra I have three devices always on managing a few dozen users every day and they work fine. They reboot every now and then, but I'm not sure if it's due to the power supply or temperature as they are not in temperature controlled spaces.

If you get these problems it might be due to the setup and pins that you are using.

Now I have a various range of ESPs and sensors, if you send me the configuration that you are using, both software and hardware I can try to replicate and see if there's a bug.

Depending on the device you cannot freely use the available pins of the ESP. Some pins have special usages and they will cause issues on esp-rfid if you use them for your sensors.

I'm going to close this for now as I don't have enough information to reproduce, but if you can replicate and provide more info I will have a look. Cheers

closed 2024-03-23 by matjack1