omersiar/esp-rfid #623: ESP-RFID Relay Board (12V for ESP8266 Board) drivers problem

MichalOmnilan, 2024-03-12:
I am trying to repeat a project but I can't send anything to my esp board. I tried a few things i found online such as  flashing the board which gave me: "error: Failed to open COM8", "error: espcomm_open failed" and "error: espcomm_upload_mem failed". In devices and printers i found COM8 with status: driver error.
Does anyone know which drivers I should download and what determins which are needed?

matjack1, 2024-03-24:
@nardev I'm not familiar with the errors reported here, can you help?

MichalOmnilan, 2024-03-26:
It seems the drivers on my pc were incorrect. I have found the right drivers here is link to the website with drivers and information about it
https://www.prolific.com.tw/US/ShowProduct.aspx?p_id=225&pcid=41

matjack1, 2024-04-09:
Excellent!

closed 2024-04-09 by matjack1