omersiar/esp-rfid #627: [FEATURE REQUEST] Do not try to reconnect to Wi-Fi when scanning "Admin" access tags in Client mode

MaximeSharp, 2024-04-09:
Using the lastest dev version,

When scanning a "Admin" access tag while the device isn't connected to the Wi-Fi network in Client mode, the device switch on the relay then try to connect to the Wi-Fi network before turning it off. 
If it failed to connect, the relay is triggered for ~10 seconds instead of the configured Activate Time (i.e 400ms).

No doubt it's a useful feature if you use the device in Access Point mode, but in Client mode it might not be necessary since it automaticaly reconnect when the Wi-Fi network is available again.

```
[ INFO ] PICC's UID: b0b4c9a3 MIFARE 1KB = known PICC
[ INFO ] User Name: '1' has admin access, enable wifi
[ INFO ] Read timeout or clean after read
mili : 39232
activating relay 0 now
[ INFO ] Trying to connect WiFi: XX
[ INFO ] WiFi BSSID: XX:XXX:XX:XXX:XXX:XXX
[ INFO ] BSSID locked
..............................
[ WARN ] Couldn't connect in time
[ INFO ] wifi | Enabling WiFi | 
54244
39232
400
0
deactivate relay after this
mili : 54321
```

matjack1, 2024-04-23:
Hey @MaximeSharp thanks for the issue! I think it wasn't intended to keep the door open for 10 seconds, it looks like a bug to me.

Have a look at this code if you can: https://github.com/esprfid/esp-rfid/pull/633 it should fix the issue. Let me know if that works, thanks!

MaximeSharp, 2024-04-23:
Hi,
I can confirm that does fix the issue ! 

I can also confirm issue #631, it still trigger the relay even if I disable the access on Tuesday  : 
[image]

closed 2024-05-09 by matjack1