omersiar/esp-rfid #631: Opening hours not working

neel8440924, 2024-04-17:
I am using build 173 with rc522 reader. Opening hours setting not working, non admin tag can activate relay and open door in non working hours(unticked hours) and log also shows access granted?

MaximeSharp, 2024-04-23:
I can also confirm issue #631, it still trigger the relay even if I disable the access on Tuesday  : 
[image]
[image]

```
[ INFO ] PICC's UID: 90535856 MIFARE 1KB = known PICC
[ INFO ] User Name: '3' has access relay
[ INFO ] Read timeout or clean after read
mili : 1006493
activating relay 0 now
1006893
1006493
400
0
deactivate relay after this
mili : 1006893
```

PR #632 seems to fix this issue

neel8440924, 2024-05-06:
I have tested this PR #632 for some time, but I don't know I think the problem is still there, I have noticed sometimes it deny access even in opening hours also, I was not able to test it fully as it was causing problem for the users so I am not using this feature for now, maybe others can test it more and report back.

matjack1, 2024-05-09:
@neel8440924 we are using this feature daily with hundreds of people and it works. The only issue that I had was with losing connection with the NTP server. If you lose that connection, then the esp can drift quite quickly and have a time very far from reality.

So if you (or someone else) want to try this feature more and if you have problems, please also attach the logs of the accesses, so that we can see at what time the access has been denied and we can check if the reason is the time drift.

Thanks!

neel8440924, 2024-05-10:
@matjack1  I have created a test setup to check it in a more controlled way and mess with settings without affecting other people. Herein this example , it denies access even in opening hours. I have attached screenshot of general setting, NTP setting, user setting as well as access logs
[image]
[image]
[image]
[image]

matjack1, 2024-06-11:
hey @neel8440924 I'm very sorry for the delay here, I've just merged in the `dev` branch some fixes to the logs that I had made. If you want to test the latest `dev` build maybe it can give you better results. Let me know if you can. Thank you very much!

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1