omersiar/esp-rfid #635: PN532 Uids missing 0 sometimes

neel8440924, 2024-05-12:
I am using fix access log build and rc522 reader and UID was scanned correctly, but every few hours rc522 would stop responding which can be fixed by restarting esp. So I tried using PN532 reader but now it misses 0s in Uids which causes some users to get locked out.

matjack1, 2024-06-11:
Hey @neel8440924 since I'm not actively using RC522 reader and the bug seems nasty, if it happens after a few hours, have you considered enabling the auto-restart of esp-rfid, so that the problem is mitigated?

matjack1, 2024-07-23:
Yes @neel8440924 I can have a look at some point soon, but if anyone has time to help on this is welcome as I don't normally use a PN532, so it's a bit cumbersome for me to test this :(