omersiar/esp-rfid #641: v2 problems with users

pvtex, 2024-06-06:
the pincode of a user is not read succesfully.
sometimes a pincode form an other user is shown.
looking in the user files in the SPIFFS filesystem the pincodes are right.

[image]

in the filesystem:

```
{"command":"userfile","uid":"???????","pincode":"8721","user":"Bjoern","acctype":99,"acctype2":null,"acctype3":null,"acctype4":null,"validsince":0,"validuntil":2145916800}
```

pvtex, 2024-06-07:
the pin that you enter in the user mask is active and can be used.
but if you reload the user list all users are showing the same pincode of one random user.

pvtex, 2024-06-07:
with debugoutputs i can see that the right pins are read in **void ICACHE_FLASH_ATTR sendUserList**

but on the way through  serializeJson and ws to the browser it get lost

pvtex, 2024-06-07:
the json opject item hold the right pincode.
the jsonarray users has  wrong pincodes.

pvtex, 2024-06-07:
think i found the solution the variable pincode was declared as const char.
const chars are used as pointers in nestedobject.
cast pincode to a string should help

pvtex, 2024-06-07:
had to change the file wsResponses.esp

```
String pincode = "";
```
and
```
if(json.containsKey("pincode"))
{
	pincode = String((const char*)json["pincode"]);
}
```

matjack1, 2024-06-11:
Thank you very much @pvtex for reporting this bug. I can confirm it.

The problem I think it's earlier than the lines highlighted in your last comment, unfortunately. I was able to reproduce the pincode corruption even by just updating manually a few of the pincodes. I'm going to fix this surely before v2 is done.

Sorry for this and thank you very much for reporting.

matjack1, 2024-06-11:
You were right @pvtex !

I've applied your fix and that's fine :) I'm going to close this as the fix is in `dev`.

Thank you :+1:

PR #645 (2024-06-11): Fix pincode reading in websocket
This was affecting the user list and the user export.

Fix provided by @pvtex, thank you very much!

Fix #641 and #608
closed 2024-06-11 by matjack1