omersiar/esp-rfid #646: cant build generic from dev branch

pvtex, 2024-06-12:
i cant build generic image from dev branch.
this is the failure:

in rfid.esp in function genericRead() there is a Serial read in line 176.
this is no problem in debug build, because the Serial is set up.
but in generic build the Serial is not setup and the Serial avaiable and read are leading to a compile error.

matjack1, 2024-07-16:
Hey @pvtex thank you for this report. I can successfully build, but I feel like you are right. This code that reads from serial has been there for a while though, so I'm not sure what has changed recently.

How are you building your project? I'm using Platformio and everything works well for me. Also the automatic build is working, so maybe there's something particular with your build process?

pvtex, 2024-07-16:
I am using platformio with visual code.
All on the newest version.
Even the included libraries are the newest version.
Perhabs there was a change.

pvtex, 2024-07-23:
But to make it failure proof that serial write should be there only i. Debug mode.
So put a ifdef DEBUG around it

matjack1, 2024-07-23:
The problem is that some sensors are reading through the Serial, so it should stay there. I'll try that sensor soon and see if that works.

matjack1, 2024-09-21:
I've just merged this in dev: https://github.com/esprfid/esp-rfid/pull/655 that should sort out this problem :)

closed 2024-09-21 by matjack1