omersiar/esp-rfid #649: Hikvision Wiegand Keypad: ESP-RFID recognises digits partially

digiwolf, 2024-07-16:
Hi – I bought a Hikvision Wiegand 26/34 Keypad (DS-K1802MK) to connect with ESP-RFID (v.2.0.0rc) however it only recognises input digits partially. Is this a known bug or are some Wiegand devices compatible whereas others aren't?

`{
  "command": "configfile",
  "network": {
    "bssid": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "ssid": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "wmode": 0,
    "hide": 0,
    "pswd": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "offtime": 0,
    "dhcp": 1,
    "ip": "",
    "subnet": "",
    "gateway": "",
    "dns": "",
    "apip": "192.168.4.1",
    "apsubnet": "255.255.255.0"
  },
  "hardware": {
    "readertype": 1,
    "wgd0pin": 13,
    "wgd1pin": 12,
    "sspin": 0,
    "rfidgain": 32,
    "wifipin": 2,
    "doorstatpin": 4,
    "rtype": 1,
    "ltype": 0,
    "rpin": 5,
    "rtime": 2000,
    "mqttlocktopic": "",
    "mqttlockmsg": "",
    "openlockpin": 255,
    "maxOpenDoorTime": 0,
    "doorbellpin": 255,
    "accessdeniedpin": 255,
    "beeperpin": 15,
    "ledwaitingpin": 14,
    "useridstoragemode": "decimal",
    "removeparitybits": false,
    "numrelays": 1,
    "relay2": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay3": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "relay4": {
      "rtype": 1,
      "ltype": 0,
      "rpin": 4,
      "rtime": 400
    },
    "requirepincodeafterrfid": false,
    "allowpincodeonly": true,
    "doorname": "Door"
  },
  "general": {
    "hostnm": "esp-rfid-apt3",
    "restart": 86400,
    "pswd": "admin",
    "openinghours": [
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111",
      "111111111111111111111111"
    ]
  },
  "mqtt": {
    "enabled": 1,
    "host": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "port": 1883,
    "topic": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "user": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "pswd": "xxxxxxxxxxxxxxxxxxxxxxxx",
    "autotopic": false,
    "syncrate": "300",
    "mqttlog": 1,
    "mqttha": 1
  },
  "ntp": {
    "server": "pool.ntp.org",
    "interval": 30,
    "timezone": 0,
    "tzinfo": "EET-2EEST-3,M3.5.0/03:00:00,M10.5.0/04:00:00"
  }
}`

matjack1, 2024-07-23:
Hello @digiwolf if you are reading between 24 and 34 bits it should work with whatever Wiegand compatible device. Can you please share some logs of what's happening?

A few troubleshooting ideas, are you using the same ground between the ESP and the Wiegand sensor? This is mandatory. Then have you tried inverting D+ and D- of the Wiegand, maybe you have configured them in the opposite way?

Apart from that, if you could build the debug version and send some console logs of what's happening when passing a card I might be able to help.

matjack1, 2026-10-01:
Closing this sorry. Meanwhile we have released v2, so if you are still interested, try with the new version and reopen if it's still valid

closed 2026-10-01 by matjack1