omersiar/esp-rfid #650: esp-rfid reboots when logs are accessed.

FOSSBOSS, 2024-08-06:
Not sure yet if it is just me, or if it is everyone using this project.
When I try to access the logs, I get :
"Please wait while fetching data."
no amount of waiting helps, it never returns the logs. 

an additional issue i have is that from the "logs" section, access, or event, 
the link to the status section doesn't work. Ill have a look at the code there in a minute.

esp-rfid Access Control  1.0.2

matjack1, 2024-08-14:
Hey @FOSSBOSS can you try building the `dev` branch? It should be much better.

I'll have a go hopefully soon in releasing a new v2 release candidate as we should be close to release now!

matjack1, 2024-09-21:
Hey @FOSSBOSS I've just release a new release candidate for V2 here: https://github.com/esprfid/esp-rfid/releases/tag/2.0.0.rc1

I hope it will work better, let me know if you can! :)

closed 2024-09-21 by matjack1

FOSSBOSS, 2024-11-02:
I will check that out! Thanks!