omersiar/esp-rfid #667: Users page error

hsmofm, 2024-11-23:
With Multiple pages of users created, the page stops loading, users stops been approved on door, cannot backup users config

matjack1, 2025-02-04:
hey @hsmofm are you using v2 of esp-rfid? How many users do you have? Also, have you tried waiting for a fair amount of time? The browser will try again and at some point it might work.

The MQTT API is more reliable than the web UI. Unfortunately the websocket library has some limitations on ESP8266 that I've not been able to solve

k1n6b0b, 2025-07-16:
So Im just testing this out for our makerspace, we have 161 users and loading the user page crashes (reboot) each time. Wondering if there's any update here, forks or testing i could check out. If a limitation of the hardware, has a separate management app been considered? (We will have multiple doors, so i was thinking along this line anyways if its possible)

matjack1, 2025-07-27:
Hello @k1n6b0b I'm familiar with the bug you are reporting :(

I've been digging a lot on the problem, but from what I understood it's a memory issue in the library ESPAsyncTCP. I've read several reports and apparently the more data you have, less memory you have available and more likely the memory gets corrupted, leading to a reboot.

MQTT works fine to me and the web UI is now retrying, so if you are patient enough it will take a few reboots but it should work. We are using esp-rfid in our Fablab with 200+ users and it eventually works. Let me know your experience!