omersiar/esp-rfid #669: Settings Are Not Saved

unmanagedtn, 2025-02-01:
Having trouble getting it to work as it will not save the settings at all... can connected but cant save any passwords or settings....

A. Description of the hardware you are using
Wemos D1 Mini v4 (clone?) ESP8266EX 
B. The version of the software that you are using
2.0 
C. exact steps to reproduce the issue
just flashed the firmware 

Thanks 

Unmanaged615

matjack1, 2025-02-04:
Hello @unmanagedtn unfortunately I don't have the exact device that you have, I cannot reproduce what you are getting :(

If you can, would be useful if you could use the debug version and report what's happening in the serial console. On the other hand, would be also useful to see what's happening in your browser's console. If you can provide more info, maybe I can see something that is not working as expected.

unmanagedtn, 2025-02-13:
I did not forget about this I have ordered some new readers from the jungle site and will test that to be 100% sure and will report ...

unmanagedtn, 2025-02-13:
```
[20:53:42];l[20:53:42][ INFO ] ESP RFID v2.0.0
[20:53:42]Flash real id:   0016405E
[20:53:42]Flash real size: 4194304
[20:53:42]
[20:53:42]Flash ide  size: 4194304
[20:53:42]Flash ide speed: 40000000
[20:53:42]Flash ide mode:  DIO
[20:53:42]Flash Chip configuration ok.
[20:53:42]
[20:53:42][ WARN ] Failed to open config file
[20:53:42][ INFO ] ESP-RFID is running in Fallback AP Mode
[20:53:42][ INFO ] AP IP address: 192.168.4.1
[20:53:42][ INFO ] AP SSID: ESP-RFID-55477a
[20:53:42][ INFO ] sys | System setup completed, running | 
[20:53:52][ WARN ] NTP sync failed
```

unmanagedtn, 2025-02-13:
Could be the same as [this](https://github.com/esprfid/esp-rfid/issues/602)

matjack1, 2025-02-13:
hey @unmanagedtn looks like the config file is not saved on the device. Can you try using the browser inspector and checking the messages that pass on the websocket?

While you save, check also the serial output of esp-rfid when in debug mode, so that we can see what happens from the browser and from the device. Thank you so much!

unmanagedtn, 2025-02-14:
Ok so I got it to work but am confused ... in your docs you list that the pn532 uses the SPI bus but I also see a setting for i2c in hardware ? so what is the correct way to connect ? if I hook it up via spi the unit never boots until I remove the dupont cables ...

unmanagedtn, 2025-02-16:
so when the 532 is attached it is getting stuck in a boot loop now ...

unmanagedtn, 2025-02-20:
Got it to boot it was an error on my part by pulling a pin down... Still can't see the reader have ordered more to try

closed 2025-02-20 by unmanagedtn

matjack1, 2025-02-23:
hey @unmanagedtn it should use SPI and pins should be the ones shown in the README file. Have you tried those?

If you still have problems let me know, I had it working at some point. It's not my usual hardware setup but I can give it a go at some point if you need it and it doesn't work